trezor/trezor-firmware #2037: [docs]

brianddk, 2021-12-24:
**Describe the bug**
The [reproducible-build](https://github.com/trezor/trezor-firmware/blob/core/v2.4.3/docs/common/reproducible-build.md) doc describes how to make local builds and official builds produce comparable hashes, but it doesn't describe how to take a hash of the Trezor-T firmware to match the `fingerprint` field of the `releases.json` file.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the official Trezor-T firmware: `wget https://data.trezor.io/firmware/2/trezor-2.4.3.bin`
2. Zero out the sig: `dd if=/dev/zero of=trezor-2.4.3.bin bs=1 seek=5567 count=65 conv=notrunc`
3. Take the hash: `sha256sum trezor-2.4.3.bin`
4. Observe the hash value: `f75eb91587be7b99ae94ee95cefadcda8f5149cabdc485f165f1086f0ad918cc`
5. Compare to official fingerprint: `a07f69d8d2065006a79c6b5636bd046496dbcb3820b41f1d604d8a4605ca4056`
6. Question how to produce the official fingerprint from official firmware.

**Expected behavior**
I'd expect to find some console procedure through the use of `dd` or `tail` to be able to produce the official fingerprint from the official firmware.  I noted that this works with the Trezor-1 firmware by simply removing the header (`tail -c +257 trezor-1.10.4.bin > trezor-1.10.4-nolegacyhdr.bin`)

**Additional context**
I'll be happy to update the doc, but I wouldn't know what commands to list.

brianddk, 2021-12-24:
OK... did some digging and reproduced the fingerprint by hashing the firmware header:

```
 (dd if=trezor-2.4.2.bin bs=1 skip=4608 count=959; dd if=/dev/zero bs=1 count=65) 2> /dev/null | openssl blake2s256
```

Now I need to make sense of what info is in the header.

closed 2021-12-24 by brianddk

brianddk, 2021-12-24:
OK... got it.  The [firmware header](https://github.com/trezor/trezor-firmware/blob/core/v2.4.3/docs/hardware/model-t/boot.md#firmware-header) contains an array of checksums (blake2s).  These are the checksums of each 128K block of firmware.  The fingerprint is the checksum (blake2s) of the checksums.

prusnak, 2021-12-24:
You can also use [`headertool.py firmware.bin`](https://github.com/trezor/trezor-firmware/blob/master/core/tools/headertool.py) to do the calculations for you.

brianddk, 2021-12-24:
In an attempt to do it all in bash, here are my final threecommands.  Requires `dd`, `xxd`, and `openssl`

```
# Checksum Blocks
for i in {1..15}; do (dd bs=128K count=1 skip=$i if=trezor-2.4.2.bin status=none | openssl blake2s256); done

# Dump the block checksums in Firmware header
xxd -ps -c 32 -s 4672 -l 480 trezor-2.4.2.bin

# Checksum Firmware header (should match fingerprint)
(dd if=trezor-2.4.2.bin bs=1 skip=4608 count=959; dd if=/dev/zero bs=1 count=65) 2> /dev/null | openssl blake2s256
```

But I agree, the `headertool.py` is much more convient.