trezor/trezor-firmware #2039: Yoroi issue after 2.4.3 update: Sending ADA to another wallet.

JaimeToca, 2021-12-27:
**Describe the bug**
After updating my Trezor Model T I have tried to send ADA to another wallet using Yoroi but I am facing some issues. First I had to confirm the transaction as always, but after that I got a message that said **"Change address payment credential is path: "m/_path_"** I approve it, and then I get **"change address stake credential is a key hash:"** at this point Trezor Model T display a red "X" I approve it and then I get "change amount XXXXX,812998 ADA to" another address I don't know. 

Is this the right behavior?

Please check out the screenshots I am adding to this issue.

**Firmware version and revision**
 2.4.3 version. Also, I downgrade to 2.4.2 and it just works fine.

**Desktop/smartphone setup (please complete the following information):**
 - Trezor Model T version 2.4.3
 - Yoroi wallet extension version
 - OS: Linux Ubuntu 
 - Browser Chrome
 - Latest Chrome version

**To Reproduce**
Steps to reproduce the behavior:
1. Send ADA to my trezor with version 2.4.2 and stake all ADA
2. Update to version 2.4.3
3. Try to send ADA to another wallet as explained in "Describe the bug" section

**Expected behavior**
Same as 2.4.2? 

**Screenshots**
https://drive.google.com/file/d/13EF7iMf3kcIwYBpw4pBjfvqM29ogzEnn/view?usp=sharing
https://drive.google.com/file/d/1EHmPSUHqKzvQUJCkxy0KcPCJj7m4fGRX/view?usp=sharing

**Additional context**
It seems that more people from the community is facing this issue
https://www.reddit.com/r/TREZOR/comments/rk2k7m/yoroi_issues_after_the_latest_trezor_model_t/

hynek-jina, 2021-12-27:
Probably related to: https://github.com/trezor/trezor-suite/issues/4669

prusnak, 2021-12-27:
Please report this issue to Yoroi wallet, so they can fix it. They are providing the non-ideal change address which triggers this warning.

closed 2021-12-27 by prusnak

prusnak, 2021-12-27:
Yoroi issue is already opened here: https://github.com/Emurgo/yoroi-frontend/issues/2575

prusnak, 2021-12-27:
@matejcik Can you please provide more info about what needs to be changed in these two issues?
* Daedalus - https://github.com/input-output-hk/daedalus/issues/2798
* Yoroi - https://github.com/Emurgo/yoroi-frontend/issues/2575

matejcik, 2022-01-03:
for future reference:

In order to skip this warning, [the following conditions](https://github.com/trezor/trezor-firmware/blob/572f3eda20960325d353930b7b32220eec8f0c93/core/src/apps/cardano/helpers/credential.py#L199-L209) must be met on all change outputs:

1. the address type is `BASE`
2. the derivation path `address_n` matches schema `m/1852'/1815'/account'/change/address`
3. the derivation path **`address_n_staking` is `m/1852'/1815'/account'/2/0` for the same account**
4. `staking_key_hash` is unset -- it should _only_ be used if the address is staking outside Trezor's control

Reported this to Yoroi; Daedalus seems to have solved theirs just by updating Connect?

mdsaizon, 2022-01-12:
Same problem here.  Yoroi support has been of no use.  I have no idea when the problem started because i haven't used the Trezor-attached Yoroi wallet in months.  I upgraded to the latest Trezor T firmware on initial launch of the Trezor suite a few days ago, then i got the problem.  I retrograded to 2.4.2, then to 2.4.1, then back to 2.4.3. Same result every time.  Anybody get any resolution to this problem?

JaimeToca, 2022-01-13:
Hello @mdsaizon, the issue is in the Yoroi side. I believe that for the next release the Yoroi team has update the trezor connect https://github.com/Emurgo/yoroi-frontend/pull/2565 and could potentially fix the problem. Perhaps you can ask [in this thread ](https://github.com/Emurgo/yoroi-frontend/issues/2575)