trezor/trezor-firmware #2041: Ethereum L2 support

yura-pakhuchiy, 2021-12-28:
Many Ethereum L2s do not introduce new tokens and use ETH as their native token (think of L2s for Ethereum as Lightning for BTC). But according to https://github.com/trezor/trezor-firmware/pull/2038#issuecomment-1002046409 this is not supported by Trezor. Some examples of L2s are Optimism, Arbitrum and Boba. First 2 are supported by Trezor because their definitions at https://github.com/ethereum-lists/chains uses alternative name for ETH (like AETH and OETH), but in fact it is the same ETH as in Ethereum mainnet, just bridged to the side chain. For Boba definitions do not have custom ticker for ETH and this prevents inclusion into Trezor. I do not think that forcing every L2 rename ETH to something else is right thing to do, rather Trezor should support this. cc @ligi

mdhz, 2022-01-20:
Please support Arbitrum, Optimism, ZKSync, BOBA and other famous L2s which are trending.

raymonddurk, 2022-01-21:
Correct me if I'm wrong but in order to use an L2 on Ethereum you need EIP712 (as is the case in ZKSync) which was just merged or other EIPs. You don't need Trezor to add support for an L2 as if it is a whole new coin, you need the ability to bridge the chain. EIP712 was merged in the T in December and the One two days ago. What work specifically needs to be done to add support for other L2s

yura-pakhuchiy, 2022-01-22:
@matejcik this information is already present in chain definition file, eg. checkout Boba definition https://github.com/ethereum-lists/chains/blob/master/_data/chains/eip155-288.json
There is `parent` section which says that Boba is L2 and parent is Ethereum mainnet.

yura-pakhuchiy, 2022-01-22:
>  user-facing network identification besides the currency symbol

Displaying chain network name (for known chain_id) and chain_id with warning about unknown network is good idea. Even with unique names for base token of the chain, chains still may have other tokens with the same name, eg. USDC on Eth, Optimism, Arbitrum and even on other non L2 chains like Polygon or AVAX.

matejcik, 2022-01-23:
> this information is already present in chain definition file

good to know! makes this issue much more tractable

matejcik, 2024-01-25:
I believe this is now automagically supported in the new loadable definitions. We are much less strict about symbol collisions now.

closed 2024-01-25 by matejcik