trezor/trezor-firmware #2043: Still cannot sign with Trezor T

astanciu, 2021-12-28:
**Describe the bug**
I thought the newly added EIP712 support would enable signing on most defi dapps with trezor t, but that doesn't seem to be the case. Simple test: go to snapshot.org and click Join on any project. If you're connected with MetaMask + Trezzor account, it will prompt for a signature. As soon as you click Sign, the following error shows up in console:
`MetaMask Message Signature: Error: Not supported on this device`

**Firmware version and revision**
2.4.3

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite Version 21.12.2 (21.12.2.14688)
 - OS: osx 12.1
 - Browser Firefox 95.0.2 (64-bit

**To Reproduce**
Steps to reproduce the behavior:
1. Go to snapshot.org
2. Make sure you connect using Metamask + a Trezzor account
3. Click on any button Join
4. See error in console

astanciu, 2021-12-28:
Side note, not sure where else to leave this comment, hope someone can fwd internally to the right place, but the community support forum is blocking new registrations. IP not allowed. Also tried with VPN from 3 diff countries ¯\_(ツ)_/¯...

prusnak, 2021-12-28:
Please update to the latest version of the Metamask which is 10.8.0.

If you are already at the latest please report the issue here: https://github.com/MetaMask/metamask-extension

closed 2021-12-28 by prusnak