trezor/trezor-firmware #2045: TROP account type identified on T1 dsiplay as BTC account

bosomt, 2021-12-30:
**Describe the bug**
Ethereum test network TROP account type is identified on T1 dsiplay as BTC account
Behavior has changed with recent change during taproot Bitcoin account name change but it still does not show Ethereum or TROP name.
Model T does not show coin name just derivation path.

**Firmware version and revision**
1.10.1 public version and 1.10.5 freeze 

**To Reproduce**
Steps to reproduce the behavior:
1. Enable ETh test account in Trezor Suite
2. Verify account address via receive tab
3. Observe Device display

**Screenshots**
1.10.5

[image]

1.10.1

[image]

prusnak, 2021-12-30:
I think this is acceptable, but I let @matejcik decide.

TEST means Testnet, not Bitcoin Testnet and Ropsten is a Testnet network.

andrewkozlik, 2022-01-14:
The difference between the two firmware versions is due to f4b8dbefb6, see https://github.com/trezor/trezor-firmware/pull/1916#issuecomment-968600667.

I agree with @prusnak's assessment in that this information indicates the path being used, which is `m/1'/**` for all testnet coins, see also https://github.com/satoshilabs/trezor-firmware/issues/114. This is intended behavior, however I can see that it might be confusing. In case of testnet coins we could get the coin shortcut from the `CoinInfo` that is provided by Suite, which would result in showing the expected shortcut. Nevertheless, doesn't seem like a big deal since it only concerns testnet coins.

bosomt, 2022-07-26:
Closing issue as not planned to fix.
Its just testnet cosmetic problem

closed 2022-07-26 by bosomt