trezor/trezor-firmware #2048: Need a feature for splitting or combining SLIP39 mnemonic phrases

smart-contract-developer, 2022-01-05:
**User Story:**
As a Trezor user I would like a way to convert existing SLIP39 Shamir Secret Shares to a BIP39 seed and vice versa. The workflow would be similar to performing a simulated recovery to check the backup. 

There could be two new buttons below the "Check backup" button in the settings:
> 1. Combine Shamir Secret Shares into a recovery seed.
> 2. Generate Shamir Secret Shares from recovery seed.

**Specific Background Story:**
The use case here is that I initialized my Trezor Model T using the 20 word Shamir Secret Share option, then later decided I wanted a 12 word seed that I'd have some hope of memorizing if I didn't have access to a backup. Migrating to a newly generated 12 word seed is not an option for me right now because of staking obligations and pending air drops.

This post refers to a web application that can do the conversion, however I'm not completely comfortable with the security risk. Even the idea of running it locally is concerning.
https://www.reddit.com/r/BitcoinDiscussion/comments/l2r76r/convert_any_1224_seed_into_a_shamir_backup
https://iancoleman.io/slip39
https://github.com/iancoleman/slip39 - "A tool for splitting and combining SLIP39 mnemonics phrases."

If there's already a way to do this using the Trezor command line interface, please let me know. I wasn't able to find such a feature in the documentation.

prusnak, 2022-01-05:
You should never convert BIP39 seed into SLIP39 shares. In case your setup is compromised, you will never know whether you failed to protect your original BIP39 seed or the new SLIP39 shares.

Also you are not supposed to memorize your seed.

closed 2022-01-05 by prusnak

smart-contract-developer, 2022-01-05:
I said "a 12 word seed that I'd have some hope of memorizing **if I didn't have access to a backup**". I did not say I wanted to memorize a seed instead of writing it down.

_"You should never convert BIP39 seed into SLIP39 shares. In case your setup is compromised, you will never know whether you failed to protect your original BIP39 seed or the new SLIP39 shares."_
I use a sufficiently strong passphrase that I only enter on the device. I'm not worried about seeds getting compromised.

_"you will never know whether you failed to protect your original BIP39 seed or the new SLIP39 shares"_
The product lets you choose either of these as valid options, so user should be able to change their mind after the fact as they adjust to the learning curve.

matejcik, 2022-01-05:
> The use case here is that I initialized my Trezor Model T using the 20 word Shamir Secret Share option, then later decided I wanted a 12 word seed

SLIP-39 and BIP-39 backups are not designed to be converted between each other.

It is impossible to convert a SLIP-39 backup into a BIP-39 seed.

It is possible to pick a BIP-39 seed and a particular passphrase, and convert that into a set of 59-word SLIP-39 shares, optionally also with a passphrase. However, using a different passphrase with the original BIP-39 seed will yield a wallet that doesn't exist anywhere within the corresponding SLIP-39 backup -- and vice versa, using a passphrase other than the originally configured one on the SLIP-39 version will yield a wallet that cannot be obtained from the BIP-39 seed.

I am aware that a lot of people are asking for convertibility. That is however **contrary to the security assumptions of hardware wallets**, and that is one of the reasons it was not a design goal of SLIP-39.

> **_"you will never know whether you failed to protect your original BIP39 seed or the new SLIP39 shares"_**
> The product lets you choose either of these as valid options, so user should be able to change their mind after the fact as they adjust to the learning curve.

You misunderstand the point of this comment. BIP-39 and SLIP-39 offer different backup strategies. Switching the backup format also implies switching the backup strategy. However, once a backup exists, it cannot be revoked -- you now have two independent backups, each of which can be compromised.

The correct way to switch your backup strategy is to destroy the original keys and move everything to the new one.

smart-contract-developer, 2022-01-05:
This is something I've been fretting about for a year, so I appreciate the more thorough explanation of why it won't work.