trezor/trezor-firmware #2053: Sign tarball releases to PyPI

hegjon, 2022-01-06:
As a package maintainer for python-trezor for Fedora Linux [1] I would encourage signing the source code that we use build RPM packages.

python-libusb1 [2][3] can be used as an example.

Related to #1915 

[1] https://src.fedoraproject.org/rpms/python-trezor
[2] https://src.fedoraproject.org/rpms/python-libusb1/blob/3c96535e1b14a01fb9917be25a6131f307df6585/f/python-libusb1.spec
[3] https://pypi.org/project/libusb1/

matejcik, 2022-01-06:
Ah, I see, PyPI still distributes them but they're not displayed and are considered deprecated.... unfortunately, there is no obvious non-deprecated replacement. I was considering minisign but ... :woman_shrugging:

hegjon, 2022-01-06:
Yes, I had too look for them for a while until I just followed the documentation from python-libusb1 documentation [1].

Who have deprecated the distribution of .asc files?

```
jonny@fedora ~/fedora-scm/python-libusb1{12.16.1} $ spectool --list-files python-libusb1.spec 
Source0: https://files.pythonhosted.org/packages/source/l/libusb1/libusb1-2.0.1.tar.gz
Source1: https://files.pythonhosted.org/packages/source/l/libusb1/libusb1-2.0.1.tar.gz.asc
Source2: gpgkey-python-libusb1.gpg
```

[1] https://github.com/vpelletier/python-libusb1/blob/5bc97a163ee1ca98ca6bfc11045f5c4ab94ec654/KEYS

matejcik, 2022-01-06:
> Who have deprecated the distribution of .asc files?

see links from https://stackoverflow.com/a/62278202/222189

the status seems to be that the whole thing should go away but a replacement is not ready so it's grudgingly kept around