trezor/trezor-firmware #2055: Add plausible deniability to wipe-code interface

brianddk, 2022-01-10:
Currently if I am under duress and have to provide my duress-pin (wipe-code) to an attacker, they are presented with the message

```
You have entered the wipe code.  All private data has been erased

Please unplug the device
```

Obviously if I give that wipe code to an attacker they will be none too happy after seeing the wipe-code confirmation.

I suggest allowing a 128 byte wipe-code confirmation message that the owner can set at the time they set the wipe-code

```
Usage: trezorctl set wipe-code [OPTIONS]

  Set or remove the wipe code.

  The wipe code functions as a "self-destruct PIN". If the wipe code is ever entered into any PIN entry dialog, then all private data will be immediately removed and the device will be reset to factory defaults.

Options:
  -m, --message User supplied wipe-code confirmation message for plausible deniable (max 128 bytes).
  -r, --remove
  --help        Show this message and exit.
```

So my personal message could be something like

```
trezorctl set wipe-code --message "ERROR id-10t: NAND at-rest corruption detected, please perform full device recovery"
```

That way, If I ever had to provide my duress PIN, I could also give them my duress seed-mnemonic.  And providing a custom message ensures that it will not be searchable on stackexchange which would defeat the purpose.

brianddk, 2022-01-10:
Upon further contemplation, the custom message would have to be retrievable WITHOUT decrypting the firmware, so housing 128 bytes in non-encrypted storage might be difficult.  The message could be encrypted with the wipe-code PIN, but that still means two sections of firmware will need to be carved out.  256 bytes of encrypted wipe code messaging, then the rest for the real firmware encrypted with the real-pin / sd-salt.

europound, 2022-05-03:
a very good idea, or even better do not do nothing just silently erase seed and move on to main screen as usual. make it natural as possible

matejcik, 2022-11-01:
> And providing a custom message ensures that it will not be searchable on stackexchange

At the same time, by not being part of the firmware source code, this indicates that this is a _custom message_, and the only way to trigger a RSOD with a custom message is by using a wipe code.

> do nothing just silently erase seed and move on to main screen as usual

Except the main screen is now a wiped Trezor. How did you enter PIN into a wiped Trezor anyway? That's not possible unless it was actually a wipe code.

Let's say that instead we generate a new random mnemonic under the hood. This takes time which can be distinguished from just entering the right PIN and decrypting the storage -- especially if you want to be thorough about it and store the same mnemonic, so that the attacker, using your "PIN" at a later time will enter the same empty wallet.

Even if that goes through, your new wallet has no history. Are you withholding a passphrase that the attacker will now try to extract out of you?

We have considered multiple options and none of them were satisfactory in all (or even the major) scenarios. This is not entirely a technical problem, the larger component is the social one.

In terms of personal safety, the best plausible deniability option remains to set up a decoy passphrase that holds some amount of funds that is interesting to an attacker, while keeping the "main" passphrase with the majority of your savings secret.

closed 2024-03-13 by prusnak