trezor/trezor-firmware #2064: Document all the CI jobs in .yml files and autogenerate docs

grdddj, 2022-01-12:
Currently we have very little documentation on `CI`, mostly for the `build` and `test` stage (here https://github.com/trezor/trezor-firmware/blob/master/docs/ci/jobs.md) - a lot of jobs are undocumented.

We could put desciptions in the `.yml` test files themselves, connected with all the tests, and then somehow extract these things into `jobs.md` file.

Possibility:
```yml
## Build of our cryptographic library, which is then incorporated into the other builds.
crypto build:
  stage: build
  <<: *gitlab_caching
  ......

## Build of Legacy into UNIX emulator. Use keyboard arrows to emulate button presses.
## Bitcoin-only version.
legacy emu btconly debug build:
  stage: build
  <<: *gitlab_caching
  ........
```

Then having some script to go through all job definitions (`crypto build`, `legacy emu btconly debug build`, etc. - the only things in file that are not indented), grabbing the comment above them and writing it into a markdown file

closed 2022-03-18 by matejcik

bosomt, 2022-03-18:
anything needed on QA side ?

matejcik, 2022-03-18:
no.