trezor/trezor-firmware #2094: Info on ssh test vectors

greenaddress, 2022-01-28:
I've seen some (old) tests for trezor ssh authentication here: 

https://github.com/trezor/python-trezor/blob/ca45019918bc4c54f1ace899a9acf397c8f4d92f/tests/test_msg_signidentity.py#L51 with the mnemonic used in the test defined in https://github.com/trezor/python-trezor/blob/ca45019918bc4c54f1ace899a9acf397c8f4d92f/tests/common.py#L17

They use as input extracted from above:
```
# mnemonic: 'alcohol woman abuse must during monitor noble actual mixed trade anger aisle'
# URI  : ssh://satoshi@bitcoin.org
# hash : 5fa612f558a1a3b1fb7f010b2ea0a25cb02520a0ffa202ce74a92fc6145da5f3
# index: 47
# curve: nist256p1 
```

and the results, path, public key  hex and signatures hex extracted from the above:
```
path -> : 'm/2147483661/4111640159/2980290904/2332131323/3701645358'

pubkey -> '03cebfae5359d6c48b8dcf9da22b2113096548407ce21da8ab28a886f750f217f4'

signature -> '00122463a8430b74b5d8c41d7c9bacc65f0eb51ceda71b9fec112e76bf2e56d8a64a66b8e019678315dc08e3be96905ea7718ec3b731e8e57e1613671ee91d1706'
```

However, in current master of this repo this test is a bit different:

https://github.com/trezor/trezor-firmware/blob/baffe8fdf313c054d48b2648f3d137632ef8ca57/tests/device_tests/misc/test_msg_signidentity.py#L75 and mnemonic defined in https://github.com/trezor/trezor-firmware/blob/baffe8fdf313c054d48b2648f3d137632ef8ca57/tests/common.py#L30

It uses the same parameters:
```
    # mnemonic: 'alcohol woman abuse must during monitor noble actual mixed trade anger aisle'
    # URI  : ssh://satoshi@bitcoin.org
    # hash : 5fa612f558a1a3b1fb7f010b2ea0a25cb02520a0ffa202ce74a92fc6145da5f3
    # index: 47
    # curve: nist256p1 
```

but the output is a bit strange cause the path is identical (meaning that the identity is identical) but both the public key and the signature are different.

```
    # path : 'm/2147483661/4111640159/2980290904/2332131323/3701645358'
    # pubkey: '0373f21a3da3d0e96fc2189f81dd826658c3d76b2d55bd1da349bc6c3573b13ae4'
    # signature -> '005122cebabb852cdd32103b602662afa88e54c0c0c1b38d7099c64dcd49efe908288114e66ed2d8c82f23a70b769a4db723173ec53840c08aafb840d3f09a18d3'
```

I've tried to follow the changes since ca45019918bc4c54f1ace899a9acf397c8f4d92f of the repo of python-trezor up to the deprecated trezor repos as well as this one but I didn't manage to find the matching code changes to the test that caused this change in results.

Could you help me find what caused these changes or point me where my mistake is in assuming the test input are identical? thank you!

greenaddress, 2022-01-28:
I think I figured it out: https://github.com/trezor/trezor-crypto/commit/0bc1b70c4aa32b730da41ca16e3b11f570e49ffc#diff-f5a8e7f7a8dad248ca42d787fa901ce951cfe9b661c0a4c6815e9eec669c00db

The change is in the bip32 code, the node from seed used "Bitcoin seed" before and now it uses "Nist256p1 seed"

Didn't verify yet but seems that would do it.

hynek-jina, 2022-01-28:
@greenaddress So is your question still relevant?

greenaddress, 2022-01-28:
@hynek-jina yes I think I'll close this one, I still need to reproduce locally but I think the above explains the differences. Thanks!

closed 2022-01-28 by greenaddress