trezor/trezor-firmware #2128: supports for signTypedData_v4

everest2105, 2022-02-15:
Hi,
I'm trying to place a limit order on app.1inch.io - but neither with metamask nor when directly connecting Trezor I'm able to place the order:

Error message (when connecting trezor directly)
The signing of a limit order in your wallet failed. Please, use another wallet that supports the method signTypedData_v4

I thought it has to do with https://github.com/trezor/trezor-firmware/pull/1970 but I am on the current firmware 1.10.5.
(using T1)

Am I missing something? Any help is very much appreciated.
Thanks

prusnak, 2022-02-15:
> but neither with metamask nor when directly connecting Trezor

To support signTypedData_v4 in Metamask, the Metamask people have to merge this PR https://github.com/MetaMask/eth-trezor-keyring/pull/117 and deploy new version.

What do you mean by "directly connecting"?

everest2105, 2022-02-15:
the DEX 1inch also allows to connect the Trezor Wallet directly https://help.1inch.io/en/articles/5575862-how-to-connect-the-trezor-hardware-wallet-to-1inch-network - without requiring Metamask...

prusnak, 2022-02-15:
> the DEX 1inch also allows to connect the Trezor Wallet directly

I understand. This is probably missing feature in 1inch - they assume signTypedData_v4 is not supported by Trezor, but it is and they simply have not implemented it yet. Please open an issue in their issue tracker.

closed 2022-02-15 by prusnak

everest2105, 2022-02-15:
Thanks! will do