trezor/trezor-firmware #2130: Do not initialize touch controller in firmware

matejcik, 2022-02-17:
Touch controller comes pre-initialized from bootloader. I believe we don't need to re-initialize in firmware's `main.c` (doing so incurs ~500 ms delay waiting for the controller to power down and boot back up).

Is there any reason not doing the initialization would be a problem? @hiviah @prusnak 

The only thing comes to mind is, bootloader might be using different configuration. Even in that case, it should be sufficient to (a) check that the controller is up and (b) send the configuration commands.

prusnak, 2022-02-17:
> Is there any reason not doing the initialization would be a problem?

I can't think of any.

matejcik, 2022-06-09:
QA instructions:
Just install this firmware and try to interact with it, unplug and replug a couple times. See if there are any touch issues (different than with regular firmware :) )

bosomt, 2022-06-13:
**QA OK**

**Info:**
 - Suite version: desktop 22.6.2 (f45c75705f153ec0dc44cb72fec2e5ec4a031b7c)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.6.2 Chrome/94.0.4606.81 Electron/15.3.0 Safari/537.36
 - OS: MacIntel
 - Screen: 1680x1050
 - Device: model T 2.5.2 regular (revision c3b82fd651facd11ffdb481da4c64ff864b6ef7a)
 - Transport: bridge 2.0.31

matejcik, 2022-07-21:
we might want to set up configuration of the touch controller, ideally without restarting it.

this will prevent problems in case future bootloaders don't set the same parameters that we want.

PR #2396 (2022-07-26): Fix firmware initialization
This PR partially reintroduces touch controller initialization in firmware:
- turns out that if we want to change operation mode of the controller in future, we might have issues when settings from bootloader being incompatible with algorithm used in firmware - this could happen for example if new bootloader is used with intentionally downgraded firmware
- the removal of initialization also caused uninitialized i2c handler being used, which only accidentally didn't cause crashes due to fact that the touch_read function evaluated the uninitialized handler as being BU [... cut]
closed 2022-07-26 by TychoVrahe