trezor/trezor-firmware #2134: Firmware boot sequence in Rust

matejcik, 2022-02-18:
Due to memory layout optimization and the unimport mechanism, the contents of `main.py` are very fragile in terms of order of import statements: a wrongly placed import can significantly increase memory footprint, or in some cases even cause basic functions to fail (such as in #2129 experiments)

When the Rust UI is done, it should be relatively easy to implement `boot.py` fully in Rust, and that paves the way to implement all of `main.py` in Rust. This would remove the fragility and significantly shorten boot-up time. (which is currently ~3 seconds until the user sees _any_ content on screen).

In order to run the main boot sequence, we would also need to modify `modtrezorio-usb-*` and directly export the interface objects from Rust.

Open question: if the main sequence runs from Rust, when does the MicroPython interpreter start? A nice answer would be to restart it after every workflow, but we could only do that if the startup of the interpreter itself is fast enough.

jpochyla, 2022-02-23:
I've thought about this a bit, and my plan was to:

- make a Usb module for Rust
- connect it to the already-done wire codec support
- possibly: figure out if we want to have native Protobuf in the Rust code at this stage, and if so, write another codegen so we can use existing lowlevel PB primitives we already have
- the fun part: rework existing boot+workflow sequence from import-based to external. The python loop that imports a module for the new session would be replaced by a loop in Rust, that would keep a uPy interpreter ready and execute the file, much like we execute main.py now. In the beginning, it could run the current boot.py, but it opens a way to RIIR cleanly using the Usb/Pb support from above.
- now the boot sequence is in Rust, fulfilling this task.
- even more fun part: now that we actually loop for a new workflow in Rust, pull the wait for the initial msg out into this loop. Now it's possible to do a pure-Rust wf.

matejcik, 2022-02-23:
> * the fun part:

I envision this part of the sequence a little differently:

1. do a small event loop in Rust that can dispatch USB and UI events (to Rust code)
2. make sure all the relevant parts of `boot.py` are either already in Rust, or implementable in Rust
    * Lockscreen UI element
    * PIN entry element
    * storage initialization and data retrieval calls (before fully replacing `storage`, we can call C's `storage_get()` directly)
    * display init and backlight control
3. from C `main()`, invoke this Rust event loop
4. when it exits, proceed to invoke MicroPython interpreter on `main.py` (which now excludes `import boot` and friends)

next step from that is expanding the Rust part to initialize the uPy interpreter, then to run the main.py loop, etc.

jpochyla, 2022-02-23:
That's a faster way to have the boot in Rust, yes.