trezor/trezor-firmware #2139: TT shows TADA as ADA

wendys-cats, 2022-02-21:
**Describe the bug**
When signing transactions (coin or token), device incorrectly shows testnet cardano coin (ATADA) as mainnet cardano coin (ADA) with network "unknown"

**Info:**
 - Suite version: web 22.3.0 (7f59e4228578fbdad89b740957865840a7d93f9c)
 - Browser: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/98.0.4758.82 Safari/537.36
 - OS: Win32
 - Screen: 1920x1080
 - Device: model T 2.4.3 regular
 
 
**To Reproduce**
Steps to reproduce the behavior:
1. Enable Test Cardano in Suite.
2. Create Test Cardano transaction.
3. Observe information on device.

**Expected behavior**
TADA will be displayed instead of ADA, and network will be known. 

**Screenshots**
[image]
[image]

wendys-cats, 2022-02-21:
@matejcik here you go. It's not the same transaction, but that shouldn't matter, right? I'm just not sure if it's FW or connect issue.

```
[
  {
    "path": "m/1852'/1815'/0'/1/0",
    "prev_hash": "ecaa1f9786b3542ada5bc013d6b810e049428b364eae9eb3fd090908f4acbeeb",
    "prev_index": 1
  },
  {
    "path": "m/1852'/1815'/0'/0/0",
    "prev_hash": "5a250bbe95bb8bdcdc7b14f127f55d0d528ab08471d8e4561818c244d3f410c4",
    "prev_index": 0
  }
]
Outputs
[
  {
    "address": "addr_test1qrmu32ju4mns5l47k9s8p96km74pm65ew8csu5nfudenx6grraf26s66z4z6f950h8dq63k3g2j3qx9wnmx0jr2xy4qsjmnv0s",
    "amount": "2000000"
  },
  {
    "addressParameters": {
      "path": "m/1852'/1815'/0'/1/1",
      "addressType": 0,
      "stakingPath": "m/1852'/1815'/0'/2/0"
    },
    "amount": "2000288",
    "tokenBundle": [
      {
        "policyId": "21c3e7f6f954e606fe90017628b048a0067b561a4f6e2aa0e1aa6131",
        "tokenAmounts": [
          {
            "assetNameBytes": "56616375756d73",
            "amount": "1"
          }
        ]
      }
    ]
  }
]
```

matejcik, 2022-02-21:
so there's two things

(a) TADA instead of ADA -> firmware never at any point had the capability to show anything other than "ADA". This is not a straightforward fix, but we'll see what we can do.

(b) Network: unknown -> unfortunately the Detail doesn't tell the whole story here. Hopefully 5binaries will know more.

wendys-cats, 2022-02-21:
Thanks, I've already sent the issue over to them as well and hopefully they'll be able to help out. I know it's a minor issue, but it would be ideal to fix it before we release ADA implementation in Suite.

mmahut, 2022-02-21:
This is because the firmware does not match the correct network magic for testnet, which is going to be fixed by https://github.com/trezor/trezor-firmware/pull/2114.

wendys-cats, 2022-02-21:
@matejcik so apparently the is already fix for the network issue [here](https://github.com/trezor/connect/pull/1046), but it is not merged yet.

sime, 2022-02-24:
@davidmisiak can you assist with displaying TADA vs ADA?

davidmisiak, 2022-02-25:
@sime I believe that adding `network_id` param to [`format_coin_amount`](https://github.com/trezor/trezor-firmware/blob/master/core/src/apps/cardano/layout.py#L87-L88) and its callees should resolve this. @matejcik is there something that I'm overlooking that makes it less straightforward please?

matejcik, 2022-02-25:
@davidmisiak i believe that's it, yes. By "not a straightforward fix" I meant that it's a bunch of places that need to propagate the network id throughout.

davidmisiak, 2022-02-25:
Alright, shall I open a PR?

matejcik, 2022-02-25:
Please go ahead. Actually feel free to push a commit to the Plutus PR

closed 2022-03-24 by hynek-jina

wendys-cats, 2022-03-25:
Henlo, are all the PRs in place for this to be tested? @slowbackspace @davidmisiak @mroz22