trezor/trezor-firmware #2151: Show source account path when signing

andrewkozlik, 2022-03-03:
When signing a transaction Trezor should show the path of the account from which the transaction inputs originate. For example:
- more technical `Spending from m/86'/0'/0'`, or
- more user friendly: `Spending from BTC Taproot account #1`.

We need to think about the UI details. Presumably it requires an extra screen. I am not sure where the screen should be located in the signing flow. At first glance it makes sense at the beginning, but if there inputs from multiple accounts, then it makes more sense at the end so that we can show the amounts being spent from each account. It would also be good if it fit together nicely with any warnings that we display during signing.

I propose we show the screen at the beginning and ignore the multi-account use-case (just display `Spending from multiple accounts` in that case).

Weakly related to https://github.com/trezor/trezor-firmware/issues/1244.

PaperHandedPuppy, 2022-05-10:
Not only would it be informational to see which account is doing the signing, not knowing which account is signing is a security risk. A corrupted host computer could send the Trezor a request to sign something on a different derivation path than what the host is telling you it's signing for.

If you restricted yourself to *only* doing safe (easily verifiable) Eth transfers for m/44'/60'/0'/0/0 which holds the majority of your funds, but allow yourself to sign/send riskier arbitrary (not easily decipherable without a decompiler) transactions for m/44'/60'/0'/0/1 which only ever sees a much smaller amount. You'd *think* you're signing an arbitrary transaction for m/44'/60'/0'/0/1 because that's what your corrupt host is showing you, but you're *actually* signing for m/44'/60'/0'/0/0 when you don't want to

Workaround: use passphrases, instead of derivation paths, to safely keep "accounts" separate from one another. the passphrase needn't be difficult to remember, it could be 'a' for account a, 'b' for account b, etc

sime, 2022-05-31:
## Designs
### Model T 

[Source](https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=2092%3A7057)

|Single account|Multiple accounts|
|-|-|
|[image]|[image]|

### Model 1

[Source](https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=2092%3A7495)

|Single account|Multiple accounts|
|-|-|
|[image]|[image]|

andrewkozlik, 2022-06-29:
Note: If we implement https://github.com/trezor/trezor-firmware/issues/2353, then the dialog should reference not only the account but also the passphrase.

Hannsek, 2023-01-24:
related/duplicate: https://github.com/satoshilabs/trezor-firmware/issues/93#issuecomment-1400482284

matejcik, 2023-02-16:
Seems the new designs take this into consideration: account info is not in front, but it's under the "(i)" at the last confirmation screen.

We might want to iterate on that but we should do that in a separate issue; I would close this when #2680 goes in

closed 2023-04-05 by Hannsek