trezor/trezor-firmware #2167: EIP712 typed data with nested data types can't be signed

bangjelkoski, 2022-03-10:
**Describe the bug**
Hello guys, this is Bojan from @InjectiveLabs. We've been trying to implement native support for Trezor for the products that we are building but we ended up having an issue when trying to sign EIP712 typed data. This issue happens on Metamask as well when we try to sign the same EIP712 message using their native Trezor integration - which just shows that this is indeed an issue on your end. 

**Firmware version and revision**
I'm using the latest firmware: `2.4.3` and I have Trezor Model T.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite installed
 - OS: MacOs
 - Browser: Chrome
 - Version: 99.0.4844.51

**To Reproduce**
Steps to reproduce the behavior:
- Try to sign the EIP712 typed data provided below using `TrezorConnect.ethereumSignTransaction`. You can even run the tests against 

**Expected behavior**
The nested types should be parsed properly and we should be able to see the nested typed data on display and the signature should be generated successfully. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
I've tested with\without `message_hash`, `domain_separator_hash`, `metamask_v4_compat`, basically every possible scenario and it still doesn't work. If you try to click on "See the full message" on the screen on Trezor you get "Firmware Error", if you click accept you - again get "Firmware Error".

Maybe you can reference my colleague's (@xlab) implementation that actually worked even with nested types.

```ts
const eip712Data = {
"types": {
    "Coin": [
        {
            "name": "denom",
            "type": "string"
        },
        {
            "name": "amount",
            "type": "string"
        }
    ],
    "EIP712Domain": [
        {
            "name": "name",
            "type": "string"
        },
        {
            "name": "version",
            "type": "string"
        },
        {
            "name": "chainId",
            "type": "uint256"
        },
        {
            "name": "verifyingContract",
            "type": "string"
        },
        {
            "name": "salt",
            "type": "string"
        }
    ],
    "Fee": [
        {
            "name": "amount",
            "type": "Coin[]"
        },
        {
            "name": "gas",
            "type": "string"
        }
    ],
    "Msg": [
        {
            "name": "type",
            "type": "string"
        },
        {
            "name": "value",
            "type": "MsgValue"
        }
    ],
    "MsgValue": [
        {
            "name": "delegator_address",
            "type": "string"
        },
        {
            "name": "validator_address",
            "type": "string"
        },
        {
            "name": "amount",
            "type": "TypeAmount"
        }
    ],
    "Tx": [
        {
            "name": "account_number",
            "type": "string"
        },
        {
            "name": "chain_id",
            "type": "string"
        },
        {
            "name": "fee",
            "type": "Fee"
        },
        {
            "name": "memo",
            "type": "string"
        },
        {
            "name": "msgs",
            "type": "Msg[]"
        },
        {
            "name": "sequence",
            "type": "string"
        },
        {
            "name": "timeout_height",
            "type": "string"
        }
    ],
    "TypeAmount": [
        {
            "name": "denom",
            "type": "string"
        },
        {
            "name": "amount",
            "type": "string"
        }
    ]
},
"primaryType": "Tx",
"sequence": "0",
"timeout_height": "8535057",
"domain": {
    "name": "Injective Web3",
    "version": "1.0.0",
    "chainId": 1,
    "verifyingContract": "cosmos",
    "salt": "1646876620490"
},
"message": {
    "account_number": "5712",
    "chain_id": "injective-1",
    "fee": {
        "amount": [
            {
            "amount": "200000000000000",
            "denom": "inj"
            }
        ],
        "gas": "400000"
    },
    "memo": "",
    "msgs": [
        {
        "type": "cosmos-sdk/MsgDelegate",
        "value": {
            "amount": {
                "amount": "100000000000000000",
                "denom": "inj"
            },
            "delegator_address": "inj17vy49gw9xnx700z8zwqqv4exl2rgdhanv75c4r",
            "validator_address": "injvaloper1w3psm8a9td2qz06s46cxss03mz5umxaxegvhhs"
            }
        }
    ]
}};

TrezorConnect.ethereumSignTransaction({
    path: "m/44'/60'/0'/0/0",
     metamask_v4_compat: true, // doesn't matter
    domain_separator_hash: "6192106f129ce05c9075d319c1fa6ea9b3ae37cbd0c1ef92e2be7137bb07baa1",
    message_hash: "c9e71eb57cf9fa86ec670283b58cb15326bb6933c8d8e2ecb2c0849021b3ef42",
    data: eip712Data
})
```

matejcik, 2022-03-10:
First of all you should be using `ethereumSignTypedData`. I am not sure what's going on here that you even get so far as Firmware Error.

matejcik, 2022-03-10:
I am however able to reproduce the problem so maybe you just mistyped the function call?

bangjelkoski, 2022-03-10:
Hey @matejcik, I'm using `ethereumSignTypedData`. I've updated the issue, I copied the wrong function name by mistake.

sime, 2022-03-10:
I can re-produce the problem when tapping Show full array. HTML is attached too.
[image]
[ethtypedsign.html.gz](https://github.com/trezor/trezor-firmware/files/8222167/ethtypedsign.html.gz)

bangjelkoski, 2022-03-10:
And [here is the PR](https://github.com/trezor/trezor-firmware/pull/1568) that @xlab did that actually had support for this as well, maybe it can help to resolve this issue quickly as I'm sure it's not something that'll take a lot of time to resolve.

matejcik, 2022-03-10:
I have a fix on the firmware side, however the signing fails later on the `sequence` key which is not present in your message.

@bangjelkoski is that a mistake on your part, or is there something specific you expect to happen when you don't provide the value?

bangjelkoski, 2022-03-10:
Here is an example object that is used as `data`
```ts
{
    "types": {
        "Coin": [
            {
                "name": "denom",
                "type": "string"
            },
            {
                "name": "amount",
                "type": "string"
            }
        ],
        "EIP712Domain": [
            {
                "name": "name",
                "type": "string"
            },
            {
                "name": "version",
                "type": "string"
            },
            {
                "name": "chainId",
                "type": "uint256"
            },
            {
                "name": "verifyingContract",
                "type": "string"
            },
            {
                "name": "salt",
                "type": "string"
            }
        ],
        "Fee": [
            {
                "name": "amount",
                "type": "Coin[]"
            },
            {
                "name": "gas",
                "type": "string"
            }
        ],
        "Msg": [
            {
                "name": "type",
                "type": "string"
            },
            {
                "name": "value",
                "type": "MsgValue"
            }
        ],
        "MsgValue": [
            {
                "name": "delegator_address",
                "type": "string"
            },
            {
                "name": "validator_address",
                "type": "string"
            },
            {
                "name": "amount",
                "type": "TypeAmount"
            }
        ],
        "Tx": [
            {
                "name": "account_number",
                "type": "string"
            },
            {
                "name": "chain_id",
                "type": "string"
            },
            {
                "name": "fee",
                "type": "Fee"
            },
            {
                "name": "memo",
                "type": "string"
            },
            {
                "name": "msgs",
                "type": "Msg[]"
            },
            {
                "name": "sequence",
                "type": "string"
            },
            {
                "name": "timeout_height",
                "type": "string"
            }
        ],
        "TypeAmount": [
            {
                "name": "denom",
                "type": "string"
            },
            {
                "name": "amount",
                "type": "string"
            }
        ]
    },
    "primaryType": "Tx",
    "domain": {
        "name": "Injective Web3",
        "version": "1.0.0",
        "chainId": 1,
        "verifyingContract": "cosmos",
        "salt": "1646906878039"
    },
    "message": {
        "account_number": "5712",
        "chain_id": "injective-1",
        "fee": {
            "amount": [
                {
                    "amount": "200000000000000",
                    "denom": "inj"
                }
            ],
            "gas": "400000"
        },
        "memo": "",
        "msgs": [
            {
                "type": "cosmos-sdk/MsgDelegate",
                "value": {
                    "amount": {
                        "amount": "100000000000000000",
                        "denom": "inj"
                    },
                    "delegator_address": "inj17vy49gw9xnx700z8zwqqv4exl2rgdhanv75c4r",
                    "validator_address": "injvaloper1w3psm8a9td2qz06s46cxss03mz5umxaxegvhhs"
                }
            }
        ],
        "sequence": "0",
        "timeout_height": "8545415"
    }
}
```

matejcik, 2022-03-10:
thanks.
can you confirm that `0x4873bf73cf22e35776d8b23a249f93f38a6d5aa8c1a121281675094f5fac64b55a3b6cf28e140930f9185156d07f171a17e06925b5cebd95a2a8761d074e43f91c` is a valid signature of this message for address `0x73d0385F4d8E00C5e6504C6030F47BF6212736A8` ?

bangjelkoski, 2022-03-10:
> thanks. can you confirm that `0x4873bf73cf22e35776d8b23a249f93f38a6d5aa8c1a121281675094f5fac64b55a3b6cf28e140930f9185156d07f171a17e06925b5cebd95a2a8761d074e43f91c` is a valid signature of this message for address `0x73d0385F4d8E00C5e6504C6030F47BF6212736A8` ?

Yes, all good from my end. When do you think this fix can be included in a new firmware release?

matejcik, 2022-03-10:
next firmware release is expected (not confirmed yet) in April

closed 2022-03-10 by matejcik

sime, 2022-03-11:
@matejcik I'm testing this is with `trezor-fw-regular-2.4.4-d9dace52.bin` and the TT screen freezes. In the video below, it was on first try. On previous attempts, it was after running the signing a second time.

https://user-images.githubusercontent.com/216917/157853023-149cd6f1-38eb-4cdb-908b-c79557b642af.mp4

I was curious as to how EIP712 works and found similar crashes when omitting some types in 2.4.3 but wanted to try the latest build before reporting them, but meanwhile found this.

matejcik, 2022-03-11:
this is a Connect issue, caused by the data you are using being incomplete. If you update your html with [this data](https://github.com/trezor/trezor-firmware/issues/2167#issuecomment-1063883612), the signing will go through

blueblack2, 2022-04-15:
> next firmware release is expected (not confirmed yet) in April

Do you have an update on when the next firmware release will be?

prusnak, 2022-04-15:
> Do you have an update on when the next firmware release will be?

20th of April

bangjelkoski, 2022-04-21:
> > Do you have an update on when the next firmware release will be?
> 
> 20th of April

Hey, are you guys still set for the release of the new firmeware?

sime, 2022-04-21:
@bangjelkoski We hit an unexpected blocker for the April 20 release. Latest expected release is May 18.

bangjelkoski, 2022-04-21:
> 

Thanks for the update!

Slokh, 2022-05-18:
@sime @prusnak Is the expected release still today?

bangjelkoski, 2022-05-18:
Yeah, they released the firmware today!

Slokh, 2022-05-18:
> Yeah, they released the firmware today!

Awesome thanks!

commit b6543470 (2022-03-10): fix(core): support structs as array elements (fixes #2167)
commit 0b9f7db7 (2022-03-10): fix(core): support structs as array elements (fixes #2167)