trezor/trezor-firmware #2173: Upgrade to isort 5

matejcik, 2022-03-11:
https://pycqa.github.io/isort/docs/major_releases/introducing_isort_5.html
https://pycqa.github.io/isort/docs/upgrade_guides/5.0.0.html

It might be possible to use one of the new isort profiles instead of our custom configuration.

AdarshRawat1, 2023-03-17:
Can I work on this issue?

matejcik, 2023-03-17:
please go ahead!

AdarshRawat1, 2023-03-18:
Could you provide some direction on where to start?

matejcik, 2023-03-18:
I would start by upgrading isort in poetry, removing the current isort config completely (in setup.cfg i think?) and replacing it by a profile. I didn't look into which profiles are available, but I vaguely remember that there's a `black` based one, so that's where I would start.

then I'd try to run the `make pystyle` task and see what changes show up. the ideal result is no changes, in which case you're basically done.

what might happen is that the new isort will not run at all, in which case you'll need to modify the invocation (in top-level makefile and in python/Makefile) so that it works again

AdarshRawat1, 2023-03-19:
Please guide me on how I can run the `make pystyle` task.

AdarshRawat1, 2023-03-20:
> This _looks like_ windows has a problem with all our files specified with forward slashes, so there's a good chance this won't work on Windows at all. If you're willing to go the extra mile, try to do it in WSL.

Noted ! Will do it in WSL.Thanks.

matejcik, 2023-03-20:
something is wrong with your setup because no arguments are passed to isort/black

first of all you should be using `poetry` (see the last section [here](https://docs.trezor.io/trezor-firmware/core/build/index.html#poetry))

second, the makefile `PY_FILES` variable is empty
try running this command, which fills it:
`find . -type f -name '*.py'   | sed 'sO^\./OO' | grep -f ./tools/style.py.include | grep -v -f ./tools/style.py.exclude`

AdarshRawat1, 2023-03-20:
@matejcik I executed this command after following all your guidance and resources.
[image]

After this I tested if other changes are remaining - 
1.
[image]

2.
[image]

Please verify !

AdarshRawat1, 2023-03-20:
I think we need to update `pyproject.toml` as well (under the style section).
@matejcik your suggestions?

matejcik, 2024-09-05:
fixed in https://github.com/trezor/trezor-firmware/pull/3219

closed 2024-09-05 by matejcik