trezor/trezor-firmware #2174: Trezor T - NEM integration, method nemGetAddress throws error for NEM testnet

cryptoBeliever, 2022-03-11:
**Describe the bug**
When we call the Trezor Connect method using Trezor T `nemGetAddress` it throws error "Invalid NEM network". 

TrezorConnect call:

```
TrezorConnect.nemGetAddress({
     path: "m/44'/1'/0'/0'/0'",
     network: -104,
     showOnTrezor: true
}).then(function(result) {
    console.log(result);
})
```
Result:

```
{
    "id": 1,
    "success": false,
    "payload": {
        "error": "Invalid NEM network",
        "code": "Failure_ProcessError"
    }
}
```

Same request using Trezor One works fine. We noticed it when upgrading from Trezor Connect v4 to v8.

**Firmware version and revision**
Trezor T - firmware 2.4.3 (doesn't work)
Trezor One - firmware 1.10.5 (works)

**Desktop/smartphone setup (please complete the following information):**
Integration with TrezorConnect 8.2.7

**To Reproduce**
Do above Trezor Connect call.

**Expected behavior**
Trezor T should export address.

prusnak, 2022-03-11:
`network: -104,`

try changing that to 

`network: 104,`

cryptoBeliever, 2022-03-11:
@prusnak 
network=104 is NEM Mainnet
network=-104 is NEM Testnet

Mainnet works fine. The problem is with Testnet (but for legacy firmware on Trezor One it works). The problem exists only on Trezor T.

prusnak, 2022-03-11:
> network=-104 is NEM Testnet

According to the following Testnet is 152, so try that value.

https://github.com/trezor/trezor-firmware/blob/d9dace5211646cfb18ca8b46308764f49146d1e2/core/src/apps/nem/helpers.py#L8

sime, 2022-03-11:
Works for that way too.

Similarly, documented here: https://github.com/trezor/trezor-suite/blob/63bcf437970495a69cdda2885120a0abfb45042f/packages/connect-explorer/src/js/data/methods/nem/common.ts

prusnak, 2022-03-11:
> Similarly, documented here: https://github.com/trezor/trezor-suite/blob/63bcf437970495a69cdda2885120a0abfb45042f/packages/connect-explorer/src/js/data/methods/nem/common.ts

Hm, it seems there is a typo there. Testnet should be 0x98, not 0x96.

cryptoBeliever, 2022-03-11:
Thank you @sime @prusnak

Yes, those are correct values (https://github.com/NemProject/nem.core/blob/fe62cae59b9068a4cfe79abf21be0427ff8a8db9/src/main/java/org/nem/core/model/NetworkInfos.java#L134): 
- 0x68 - mainnet 
- 0x98 - testnet
- 0x60 - mijin
 
It's -104 because 0x98 is signed 2's complement (https://www.rapidtables.com/convert/number/hex-to-decimal.html)
But we can easily convert that when doing requests to Trezor (`network < 0 ? 256 + network : network`) so it's not an issue.

After fixing Testnet id here: https://github.com/trezor/trezor-suite/blob/63bcf437970495a69cdda2885120a0abfb45042f/packages/connect-explorer/src/js/data/methods/nem/common.ts you can close.

Thank you for the prompt answer and help!

prusnak, 2022-03-11:
Connect-explorer change is tracked here https://github.com/trezor/trezor-suite/pull/5101

closed 2022-03-11 by prusnak

matejcik, 2022-03-11:
It's still strange that this worked for T1 but not for TT.

But that must be something on Connect side, because the negative number cannot be encoded into the protobuf message so it could not have reached Trezor.

matejcik, 2022-03-11:
Ah, I see the issue now.
T1 is passing around the network as `uint8_t` so even though on the wire the value is 0xFFFFFF98, it gets cut down to 0x98 in all places where it matters.
TT is using the full width of uint32.
I don't think we need to do anything about this on the firmware side.

cryptoBeliever, 2022-03-11:
Yes, @matejcik. Converting -104 -> 152 in Trezor requests works fine for both T1 and TT so you don't need to change anything.
Thank you for your help!