trezor/trezor-firmware #2189: Deterministic build of `bootloader-2.0.3.bin` produces `binctl` hash mismatches

brianddk, 2022-03-28:
Attempting a deterministic build of `bootloader-2.0.3.bin`.  I assumed the proper tag was `core/bl2.0.3`, and the proper build command was `PRODUCTION=1 ./build-docker.sh core/bl2.0.3 local`?  I attempted this on an updated edition of Ubuntu 21.10 (x86_64) v5.13.0-37-generic.  Here's the full build script and the conflicting outputs of `binctl` that I would have expected to match.

### Build Script

```shell
#!/bin/bash

# Build bootloader v2.0.3
#
bl_tag="core/bl2.0.3"
src="$HOME/src/trezor-firmware"
cd "$src"
git restore --staged .
git checkout -f "$bl_tag"
PRODUCTION=1 ./build-docker.sh "$bl_tag" "local"

# Download the official 2.0.3 bootloader
#
bl_prod="bootloader-2.0.3.bin"
wd_tag="05f410d"
download="$src/download"
url="https://github.com/trezor/webwallet-data/raw/$wd_tag/bootloader/2"
mkdir -p "$download"
curl -o "$download/$bl_prod" -L "$url/$bl_prod"

# Make python environment for "binctl"
#
git checkout -f "core/v2.1.1" ./python
python3 -m venv binctl.venv
source ./binctl.venv/bin/activate
python3 -m pip install --upgrade pip setuptools wheel pyblake2 ./python

# Compare released bootloader, to built bootloader
#
set -x
./tools/binctl "$download/$bl_prod"
./tools/binctl ./build-docker/bootloader/bootloader.bin
```

### Output

```
+ ./tools/binctl ./download/bootloader-2.0.3.bin
TREZOR Bootloader Image
  * magic   : TRZB
  * hdrlen  : 1024
  * expiry  : 0
  * codelen : 93184
  * version : 2.0.3.0
  * fixver  : 2.0.0.0
  * hashes: OK
    - 00 : cd0c14938dab0375c7547cf4087457f99f7762a1dcf8eb0a774b3f9ba2678058
    - 01 : 0000000000000000000000000000000000000000000000000000000000000000
    <clip>
    - 15 : 0000000000000000000000000000000000000000000000000000000000000000
  * sigmask : 0x03 = [1 2 . . . . . .]
  * sig     : 408791dbd5316b554e8159e042fa67d7afab8e4e5611d39cec0f3def40355cbcbc0d92119065c158fff67070818c57deb7b656e3af0081c234a6d8fd7c15980d
  * total   : 94208 bytes
  * fngprnt : 1b59b724943c2e4af724f3989716ecf219f4c87ba512164af1c8c413c1737af5

+ ./tools/binctl ./build-docker/bootloader/bootloader.bin
TREZOR Bootloader Image
  * magic   : TRZB
  * hdrlen  : 1024
  * expiry  : 0
  * codelen : 93184
  * version : 2.0.3.0
  * fixver  : 2.0.0.0
  * hashes: OK
    - 00 : 7abc73a7966fa3b008a86c17a6c74a0c6193e7cef64b0528e56a4b4f41a8904a
    - 01 : 0000000000000000000000000000000000000000000000000000000000000000
    <clip>
    - 15 : 0000000000000000000000000000000000000000000000000000000000000000
  * sigmask : 0x00 = [. . . . . . . .]
  * sig     : 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
  * total   : 94208 bytes
  * fngprnt : 872e8def62a146dac2a56ddcafb01247bc2642aaf29f38bf12d965988aaf850b
```

The sizes line up but the hashes don't.

prusnak, 2022-03-28:
can you send the output of the following command?

```
diff -u <(xxd ./download/bootloader-2.0.3.bin) <(xxd ./build-docker/bootloader/bootloader.bin)
```

brianddk, 2022-03-28:
Both bin files, and the diff are attached in the tarball as well as the output inline

> `diff -u <(xxd ./download/bootloader-2.0.3.bin) <(xxd ./build-docker/bootloader/bootloader.bin)`

```
--- /dev/fd/63	2022-03-28 11:13:34.493815553 -0500
+++ /dev/fd/62	2022-03-28 11:13:34.493815553 -0500
@@ -1,7 +1,7 @@
 00000000: 5452 5a42 0004 0000 0000 0000 006c 0100  TRZB.........l..
 00000010: 0200 0300 0200 0000 0000 0000 0000 0000  ................
-00000020: cd0c 1493 8dab 0375 c754 7cf4 0874 57f9  .......u.T|..tW.
-00000030: 9f77 62a1 dcf8 eb0a 774b 3f9b a267 8058  .wb.....wK?..g.X
+00000020: 7abc 73a7 966f a3b0 08a8 6c17 a6c7 4a0c  z.s..o....l...J.
+00000030: 6193 e7ce f64b 0528 e56a 4b4f 41a8 904a  a....K.(.jKOA..J
 00000040: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000050: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000060: 0000 0000 0000 0000 0000 0000 0000 0000  ................
@@ -57,11 +57,11 @@
 00000380: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000390: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 000003a0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-000003b0: 0000 0000 0000 0000 0000 0000 0000 0003  ................
-000003c0: 4087 91db d531 6b55 4e81 59e0 42fa 67d7  @....1kUN.Y.B.g.
-000003d0: afab 8e4e 5611 d39c ec0f 3def 4035 5cbc  ...NV.....=.@5\.
-000003e0: bc0d 9211 9065 c158 fff6 7070 818c 57de  .....e.X..pp..W.
-000003f0: b7b6 56e3 af00 81c2 34a6 d8fd 7c15 980d  ..V.....4...|...
+000003b0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
+000003c0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
+000003d0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
+000003e0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
+000003f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
 00000400: 0000 0110 e95c 0208 f9ec 0208 f9ec 0208  .....\..........
 00000410: f9ec 0208 f9ec 0208 f9ec 0208 f9ec 0208  ................
 00000420: f9ec 0208 f9ec 0208 f9ec 0208 f9ec 0208  ................
@@ -1765,7 +1765,7 @@
 00006e40: 895a 0308 5f5c 0308 0e5c 0308 245c 0308  .Z.._\...\..$\..
 00006e50: 0000 0408 a15c 0308 b75c 0308 f75c 0308  .....\...\...\..
 00006e60: 105d 0308 5452 5a46 2e5d 0308 465d 0308  .]..TRZF.]..F]..
-00006e70: 5167 0308 e25d 0308 f85d 0308 2046 fff7  Qg...]...].. F..
+00006e70: 5267 0308 e25d 0308 f85d 0308 2046 fff7  Rg...]...].. F..
 00006e80: 25f8 4ff4 7a70 04f0 47ff 013c 002c f5d1  %.O.zp..G..<.,..
 00006e90: bdf8 2430 9b06 03d4 fff7 48f8 05f0 e0fe  ..$0......H.....
 00006ea0: fff7 b6fc 05f0 96fc 069b 0448 1844 07f0  ...........H.D..
@@ -1938,7 +1938,7 @@
 00007910: 3b64 0308 5007 0010 c560 0308 5452 5a46  ;d..P....`..TRZF
 00007920: db60 0308 0000 0408 5807 0010 f360 0308  .`......X....`..
 00007930: c467 0308 3260 0308 6160 0308 5f5c 0308  .g..2`..a`.._\..
-00007940: cd64 0208 5167 0308 0e61 0308 6e61 0308  .d..Qg...a..na..
+00007940: cd64 0208 5267 0308 0e61 0308 6e61 0308  .d..Rg...a..na..
 00007950: 3a68 6048 521a 00eb 4310 01f1 0051 04f0  :h`HR...C....Q..
 00007960: 53fc b0f1 aa3f 40d0 5b4a 1368 002b 17dd  S....?@.[J.h.+..
 00007970: 013b 1360 0023 0593 0793 2b68 5b04 0122  .;.`.#....+h[.."
@@ -1966,7 +1966,7 @@
 00007ad0: f08f 00bf 7007 0010 0c00 0010 9465 0308  ....p........e..
 00007ae0: 5407 0010 0c5f 0308 7f61 0308 3b64 0308  T...._...a..;d..
 00007af0: 3260 0308 6160 0308 9261 0308 a761 0308  2`..a`...a...a..
-00007b00: f861 0308 5766 0308 5167 0308 70b5 264e  .a..Wf..Qg..p.&N
+00007b00: f861 0308 5766 0308 5267 0308 70b5 264e  .a..Wf..Rg..p.&N
 00007b10: 264a 3368 c2b0 0546 1221 2548 4193 04f0  &J3h...F.!%HA...
 00007b20: 39fc b0f1 aa3f 234b 28d0 1b78 8df8 0330  9....?#K(..x...0
 00007b30: ff22 0021 01a8 07f0 03f9 40f6 0113 adf8  .".!......@.....
@@ -3023,7 +3023,7 @@
 0000bce0: 0b48 f9f7 f9fb 0b48 f9f7 f6fb 02f0 ddff  .H.....H........
 0000bcf0: fee7 00bf f809 0010 7566 0308 8466 0308  ........uf...f..
 0000bd00: 8e66 0308 9866 0308 a566 0308 af66 0308  .f...f...f...f..
-0000bd10: b866 0308 c266 0308 07b5 074b 1a68 0192  .f...f.....K.h..
+0000bd10: b966 0308 c366 0308 07b5 074b 1a68 0192  .f...f.....K.h..
 0000bd20: 019a 1b68 9a42 01d0 00f0 64f8 03b0 5df8  ...h.B....d...].
 0000bd30: 04eb fff7 5db9 00bf f809 0010 30b5 184d  ....].......0..M
 0000bd40: 85b0 2b68 0393 00f0 affd 0023 c4b2 0193  ..+h.......#....
[... 59 lines cut ...]
-000166c0: 0a00 0a50 6c65 6173 6520 636f 6e74 6163  ...Please contac
-000166d0: 7420 5452 455a 4f52 2073 7570 706f 7274  t TREZOR support
-000166e0: 2e0a 005f 5f73 7461 636b 5f63 686b 5f66  ...__stack_chk_f
-000166f0: 6169 6c00 656d 6265 642f 7472 657a 6f72  ail.embed/trezor
-00016700: 6861 6c2f 636f 6d6d 6f6e 2e63 0053 7461  hal/common.c.Sta
-00016710: 636b 2073 6d61 7368 696e 6720 6465 7465  ck smashing dete
-00016720: 6374 6564 0073 6563 6661 6c73 6500 0000  cted.secfalse...
+000166a0: 3a25 640a 0066 756e 633a 2025 730a 0032  :%d..func: %s..2
+000166b0: 3331 3134 3635 6664 0072 6576 203a 2025  311465fd.rev : %
+000166c0: 730a 000a 506c 6561 7365 2063 6f6e 7461  s...Please conta
+000166d0: 6374 2054 5245 5a4f 5220 7375 7070 6f72  ct TREZOR suppor
+000166e0: 742e 0a00 5f5f 7374 6163 6b5f 6368 6b5f  t...__stack_chk_
+000166f0: 6661 696c 0065 6d62 6564 2f74 7265 7a6f  fail.embed/trezo
+00016700: 7268 616c 2f63 6f6d 6d6f 6e2e 6300 5374  rhal/common.c.St
+00016710: 6163 6b20 736d 6173 6869 6e67 2064 6574  ack smashing det
+00016720: 6563 7465 6400 7365 6366 616c 7365 0000  ected.secfalse..
[... cut ...]
 00016740: 0000 0000 0000 0000 0000 0000 0000 0000  ................
-00016750: 0006 0708 090a 0b11 1213 1415 1617 0000  ................
+00016750: 0000 0607 0809 0a0b 1112 1314 1516 1700  ................
 00016760: 0000 0008 0040 0008 0080 0008 00c0 0008  .....@..........
 00016770: 0000 0108 0000 0208 0000 0408 0000 0608  ................
 00016780: 0000 0808 0000 0a08 0000 0c08 0000 0e08  ................
```
[trezor-firmware.tar.gz](https://github.com/trezor/trezor-firmware/files/8364597/trezor-firmware.tar.gz)

brianddk, 2022-03-30:
I've attacked another build.  I get different hashes building out of the `trezor-core` repo as opposed to the `trezor-firmware` repo, but nether match the hash of the published bootloader.  Though repeated builds on each repo do produce predictable hashes.  The firmware build (2.4.3) does have matching hashes, if I use embed the published bootloader.  Embedding the built bootloader of course causes the hashes to deviate at certain points

Attached Tarball: [builds.tar.xz.gz](https://github.com/trezor/trezor-firmware/files/8376544/builds.tar.xz.gz)

## Results of reproducible build experiment

* **trezor-core/bl2.0.3/build-docker/** - Bootloader build output from `trezor-core` at tag `bl2.0.3`
* **trezor-firmware/bl2.0.3/build/** - 2.4.3 firmware built with `core/embed/firmware/bootloader` from `bl2.0.3`
* **trezor-firmware/core/bl2.0.3/build-docker/** - Bootloader build output from `trezor-firmware` at tag `core/bl2.0.3`
* **trezor-firmware/core/bl2.0.3/build/** - 2.4.3 firmware built with `core/embed/firmware/bootloader` from `core/bl2.0.3`
* **download/** - Published retail versions of BL 2.0.3 and FW 2.4.3

matejcik, 2023-04-14:
i'm kind of reluctant to dive that deep into build history :/ esp. considering the amount of work that went into reproducibly building 2.1.0

it might very well be that we won't be able to recreate the bl 2.0.3 build because we didn't have nix then. there could be something something changed version of gcc in the dockerized debian etc.

brianddk, 2023-04-21:
Sounds good.  I'll look at doing a 2.1.0 build to see how it goes

brianddk, 2023-05-23:
@matejcik , I'm trying to work this on 2.6.0.  Few questions about the current build environment.  I'm able to verify the bootloader images match with the following command:

```
core/tools/headertool.py -h ./core/embed/firmware/bootloaders/bootloader_T2T1.bin
core/tools/headertool.py -h build/core-bitcoinonly/bootloader/bootloader.bin
```

But I can't seem to find a good tool to do the same for the Legacy firmware.  The closest I can do is:

```
diff <(xxd -g 1 legacy/firmware/bootloader.dat) <(xxd -g 1 build/legacy/bootloader/bootloader.bin)
```

which of course produces more differences than similarities.

matejcik, 2023-05-23:
`legacy/firmware/bootloader.dat` and `build/legacy/bootloader/bootloader.bin` should be byte-for-byte identical because there's no signing involved.

however keep in mind that bootloaders have their own tags. if you're building off `core/v2.6.0`, you might see differences. check out `legacy/bl1.12.1` and you should be good

brianddk, 2023-05-23:
> check out `legacy/bl1.12.1` and you should be good

Thx, I should have caught that.  I attest that 1.12.1 and 2.6.0 builds reproduce as expected!

closed 2023-05-23 by brianddk

brianddk, 2023-05-24:
@matejcik I just noticed looking over my builds that for Legacy, the "correct" embedded bootloader binary is under the tag `legacy/bl1.12.1`, which is good and expected, but the "correct" embedded bootloader for Core is under the tag `core/v2.6.0` while the "correct" source is under `core/bl2.1.0`.

I've already corrected my build scripts to do a `git restore` on `bootloader_T2T1.bin` so I'm all good, but wanted to know if I should open (then close) an issue for documentation for anyone else who might stumble upon this.

matejcik, 2023-05-25:
That's an interesting observation, worth documenting somehow.
It prooobably needs to be like this because embedding a new bootloader for core is a two-step process:
1. build & tag the bootloader
2. sign it
3. add the signed binary to git
We _could_ hold off on tagging the binary until after we have the signatures, but it's kind of problematic in terms of process.

we really should have a master document about reproducibility and how to get what for what, ideally with notes like "we don't know how to build 2.0.3 anymore, please feel free to try to figure it out & tell us."

brianddk, 2023-05-25:
> we really should have a master document about reproducibility

I'm starting on something like that presently.  I'll post the repo on this thread shortly and if your interested, then I could format it as a PR.  In simple terms, it's just a large bash hashtable with the repo tags as indexes:

```
docker_build_sh_ver["core/bl2.1.0"]="core/v2.6.0"
docker_build_sh_tag["core/bl2.1.0"]="core/bl2.1.0"
embedded_bin_tag["core/bl2.1.0"]="core/v2.6.0"

```
This documents that in order to make a reproducible build for `core/bl2.1.0`, you must checkout the version of `docker-build.sh` at the `core/v2.6.0` tag, run `docker-build.sh` with the `core/bl2.1.0` argument and, compare the build to the `bootloader_T2T1.bin` at `core/bl2.1.0` tag

brianddk, 2023-05-26:
Here's my scripts I worked out to simplify the process

https://github.com/brianddk/attestation/

If you look at the [attest](https://github.com/brianddk/attestation/tree/main/trezor/attest) directory you will see the builds I have been able to successfully reproduce.  Most of the "magic" happens in the [settings.sh](https://github.com/brianddk/attestation/blob/main/trezor/settings.sh) file that contains the BASH hashtables I hinted at earlier.  You can also run the [defaults.py](https://github.com/brianddk/attestation/blob/main/trezor/defaults.py) to get the unmodified default values.  This makes it simple to `diff` to figure out what I've had to tweek.