trezor/trezor-firmware #2190: Do not convert bech32 addresses to upper-case for QR code (T1)

hynek-jina, 2022-03-28:
Follow-up from https://github.com/trezor/trezor-firmware/issues/1751 fro Trezor One.

hynek-jina, 2022-03-30:
Currently T1 shows QR code with upper-case address (ie `BC1QUQGQ44WQ0ZJH6D920ZS42NSY4N4EV5VT8NXKE4`)
[image]

While Suite shows different QR code with lower case address (ie `bc1quqgq44wq0zjh6d920zs42nsy4n4ev5vt8nxke4`)
[image]

prusnak, 2022-03-30:
Will the Taproot QR code fit on a legacy screen if the address is not upper-cased? Let's test this (Taproot addresses are longer than Segwit v0).

Midar, 2022-04-24:
Isn't this more a bug in Suite that it's *not* uppercase there? I thought one of the ideas of BECH32 was that you all-uppercase the QR code to make it smaller?

hynek-jina, 2022-08-25:
Build the custom version of model one for QA to test the readability of QR code

grdddj, 2022-08-30:
I do not have good news here, the `taproot` address in lowercase does not fit in `V3` QR code (29x29), but requires `V4` (33x33). With our screen-height of `64 px`, it is not possible to fit `33*2` pixels on the screen, therefore it uses only one pixel per module, which is really badly visible.

(Very theoretically, we could display that 66x66 px QR code when the very top and very bottom row would be missing on the screen, it could still scan well given the error-corrections of QR codes.)

Below is the screenshot with magnified (`TREZOR_OLED_SCALE=2`) emulator in `Suite`.

On a hardware device, I was not able to scan this miniature `QR code`.

[Here](https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/pipelines/626353573) is a `Gitlab` pipeline with the change, [here](https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/2952416753/artifacts/raw/trezor-fw-regular-debug-1.11.3-f3a10d8b.bin) one can download the firmware and try it on hardware.

Also what I discovered (maybe already known) is that the QR codes in `Suite` and on `device` are different, even though they encode the same address (for `taproot` and also for `segwit`)

[image]

matejcik, 2022-08-30:
> (Very theoretically, we could display that 66x66 px QR code when the very top and very bottom row would be missing on the screen, it could still scan well given the error-corrections of QR codes.)

this might actually be viable. did you try to scan it?

grdddj, 2022-08-30:
> this might actually be viable. did you try to scan it?

Yeah, I could scan it (even though it sometimes has difficulties on hardware device due to the screen flickering - moving black lines - applicable for all QR codes however - or my phone).

It is also possible to fully omit the last two rows (last module row), which is easier to do code-wise - https://github.com/trezor/trezor-firmware/commit/2dbd19418b89f73079721fe677ae429b341ef1eb

[image]

Hannsek, 2022-08-30:
@grdddj I tried it now and actually I can scan it. So I suppose it is the camera thing. But it doesn’t look good anyway.

Midar, 2022-08-31:
I'm sorry, but I seem to be missing something important here: Why would it be desirable to have the QR code in lowercase, when BECH32 was *designed* to be case-insensitive so that it can be *uppercase* in QR codes so that the QR codes can be *smaller*? It seems them being lowercase is *exactly* the problem, so why use hacks to make them fit instead of using them the way they are designed to, which is uppercase?

matejcik, 2022-09-01:
The issue is that while bech32 is _designed_ to work in uppercase, many wallets do not accept uppercase addresses, so this improves compatibility.

With that said, you have a good point: in case of Taproot addresses, we can keep the uppercase variant for better scannability, at the expense of compatibility with (nonconformant) software.

matejcik, 2022-09-08:
@grdddj could we maybe fit more data into a smaller QR code if we used less error correction? IIRC we are using the M level everywhere.
https://www.qrcode.com/en/about/version.html
Would lowering the error correction level be enough to fit the longer Taproot lowercase address into a lower version?

If not, let's go with lowercase + larger QR code + cutoff, as implemented in the PR. There should (in theory) be no loss of readability, because the QR code pixels are still the same size in terms of display pixels.

grdddj, 2022-09-09:
> @grdddj could we maybe fit more data into a smaller QR code if we used less error correction? IIRC we are using the M level everywhere. https://www.qrcode.com/en/about/version.html Would lowering [...]

We seem to be already using `qrcodegen_Ecc_LOW` - the lowest possible error correction - for `model 1` - https://github.com/trezor/trezor-firmware/blob/master/legacy/firmware/layout2.c#L778 (which makes sense, as we do not expect any display damage, we are not paper)

From the source above, I can also see that the maximum `binary`-length for `v3` is `53`, which does not satisfy `taproot` with `62` chars. Therefore, we cannot avoid `v4` in this case (with lowercase)

Midar, 2022-09-10:
How about using lowercase for older addresses, and uppercase for Taproot? Taproot is new and not supported by old, buggy wallets anyway, so wallets will need to add support for it and test it. So why not use this as an opportunity to make uppercase addresses common so that wallets fix that when they add support for Taproot? Is there even a single wallet that supports Taproot but fails on uppercase addresses?

matejcik, 2022-09-10:
@Midar what would be the point though? It will not increase scannability of the QR code.

closed 2022-09-14 by grdddj

STew790, 2022-09-20:
**QA OK**
QR code in legacy fw is showing now lower-case address for BECH32 account.

**Info:**
- Suite version: web 22.9.2 (cfeba9a3533ec7f4a5d4f31da4ac8c8b768547ed)
- Browser: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/105.0.0.0 Safari/537.36
- OS: Linux x86_64
- Screen: 1920x1080
- Device: model 1 1.11.3 Universal (revision https://github.com/trezor/trezor-firmware/commit/4bed278e80d23077676128eba8cb2478fcd31120)
- Transport: bridge 2.0.32