trezor/trezor-firmware #2195: Fix patching of library paths of old emulator binaries

mmilata, 2022-03-29:
See #2194 and https://github.com/NixOS/nixpkgs/pull/149731. As autoPatchelfHook exists for different usecase maybe consider different approach like compiling the emulators with statically linked SDL.

matejcik, 2022-03-29:
the issue is not SDL though but ld.so

prusnak, 2022-03-29:
Also this would mean that we have to recompile all old emulators, which is quite a hassle, but if we would be able to produce completely static binaries it might be worth the effort?

mmilata, 2022-03-29:
Right, I belive in some cases it's both ld.so and SDL, but only static linking won't save us.

mmilata, 2022-06-21:
For future reference: In #2344 we adapt our code to the new autoPatchelf function which does not support files as arguments, just directories. Upstream, NixOS/nixpkgs#166118 seems to bring this functionality back.