trezor/trezor-firmware #2199: trezorctl: AttributeError: 'TrezorctlGroup' object has no attribute 'resultcallback'

lestephane, 2022-03-30:
```
$ python3 --version
Python 3.8.10
$ python3 -m venv trezor
$ source trezor/bin/activate
$ pip3 install setuptools wheel
$ pip3 install trezor
$ trezorctl
Traceback (most recent call last):
  File "/home/elstephane/VirtualEnvironments/trezor/bin/trezorctl", line 5, in <module>
    from trezorlib.cli.trezorctl import cli
  File "/home/elstephane/VirtualEnvironments/trezor/lib/python3.8/site-packages/trezorlib/cli/trezorctl.py", line 191, in <module>
    @cli.resultcallback()
AttributeError: 'TrezorctlGroup' object has no attribute 'resultcallback'
```

sime, 2022-03-30:
@lestephane What operating system? Have you managed to install the trezor package previously/?

matejcik, 2022-03-30:
We are currently incompatible with `click` 8.1. As a workaround, `pip3 install "click<8.1"`

The current code in master has the right versioning set, but the published package does not.

Maybe a better approach is to fix the compatibility.

jonathancross, 2022-05-02:
@matejcik  the fix you suggested isn't working for me when using the [PyPi](https://pypi.org/project/trezor/) version of `trezorctl`:
```
$ pip3 install "click<8.1"
Requirement already satisfied: click<8.1 in /usr/lib/python3/dist-packages (7.1.2)

$ trezorctl
Traceback (most recent call last):
  File "/home/████/.local/bin/trezorctl", line 5, in <module>
    from trezorlib.cli.trezorctl import cli
  File "/home/████/.local/pipx/venvs/trezor/lib/python3.9/site-packages/trezorlib/cli/trezorctl.py", line 191, in <module>
    @cli.resultcallback()
AttributeError: 'TrezorctlGroup' object has no attribute 'resultcallback'
```

jonathancross, 2022-05-02:
Nevermind... because I was using `pipx` environment, it was still using the incorrect 8.1 version instead of the one in `/usr/lib/python3/dist-packages`  😬

prusnak, 2022-05-27:
This bit me again today. I think we should make trezorctl compatible with click 7.x, 8.0.x and 8.1.x and publish a new release on PyPI.

sime, 2022-05-31:
Can also confirm on OS X 12.2.1 with a new user the following is required to make `trezorctl` work. (purposefully tried a new user to get the virgin environment variables).

```
pip3 install trezor
pip3 install "click<8.1"
$HOME/Library/Python/3.8/bin/trezorctl version # Returns 0.13.0
```

jackinloadup, 2022-06-09:
I was able to use `trezorctl` after making the following simple modification.
```diff
diff --git a/python/src/trezorlib/cli/trezorctl.py b/python/src/trezorlib/cli/trezorctl.py
index 8e133d85c..a1f7a86fe 100755
--- a/python/src/trezorlib/cli/trezorctl.py
+++ b/python/src/trezorlib/cli/trezorctl.py
@@ -189,7 +189,7 @@ def cli_main(
 cli = cast(TrezorctlGroup, cli_main)
 
 
-@cli.resultcallback()
+@cli.result_callback()
 def print_result(res: Any, is_json: bool, script: bool, **kwargs: Any) -> None:
     if is_json:
         if isinstance(res, protobuf.MessageType):
```

Here are steps to reproduce the fix. 
```sh
git clone https://github.com/trezor/trezor-firmware.git
cd trezor-firmware
patch -p1 < patch_file_above.patch
nix-shell
poetry install
poetry shell
trezorctl list
```
This fix only works inside the nested nix and poetry shell. I didn't see any issues but haven't done any extensive testing. I didn't find instruction on running a test suite.

commit 5173cd3b (2022-06-29): fix(python): click 8.1 compatibility

fixes #2199
closed 2022-06-30 by matejcik