trezor/trezor-firmware #2205: Unable to use Trezor Model T as security key on Google with Safari

sime, 2022-04-04:
**Describe the bug**
Adding Trezor Model T as a security key to Google fails.

**Firmware version and revision**
2.4.3

**Desktop/smartphone setup (please complete the following information):**
 - OS: macOS
 - Browser Safari 
 - Version 15.3

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://myaccount.google.com/signinoptions/two-step-verification
2. Authenticate
3. Click 'Add Security Key'
4. See 'Got your security key?' modal and click 'Next'
5. Press the green tick 

**Expected behavior**
Successful pair Model T to Google account

**Screenshots**
[image]
[image]

**Additional context**
Model 1 works.

sime, 2022-05-04:
@mcudev Just to clarify, the issue is with Google.com ?

mcudev, 2022-05-14:
@sime yes, google.com. testing again today, i find that things are working a little better/differently with chrome version 101.0.4951.64 (Official Build) (64-bit).

now, i can get my key registered with the "FIDO2 Register" process and working with this google accounts + chrome combo, there is just one workaround that i had to do. when registering the key, i get this screen:

[image]

if you click allow, the process fails with:

[image]

when you click skip, you can name the key and continue on and things work. also, when logging in using the fido2 authenticate process, with this google + chrome combo, it actually lets me in instead of seeming like it works and then forcing me to use a fallback 2fa mechanism.

firefox still uses the u2f authenticate process and that still works too.

sime, 2022-05-16:
Can confirm it is failing for me on Chrome and Safari.

andrewkozlik, 2023-02-17:
Reopening. PR #2834 only fixes https://github.com/trezor/trezor-firmware/issues/2205#issuecomment-1109755886 for Chrome.

bosomt, 2023-02-19:
I can confirm that it does not work on my Safari + macOS ;)

 - Device: model T 2.5.4 Universal (revision 0a955b4ad6d9fb5e7a1dbd3333548ffd8b33399a)
 - Safari Version 16.3 (18614.4.6.1.5)

closed 2023-02-23 by andrewkozlik