trezor/trezor-firmware #2208: Expose model information and metadata from boardloader

matejcik, 2022-04-05:
In order to support the same bootloader and/or firmware image on multiple Trezor models, we need some info exposed about the current hardware.

We definitely need to see **model identifier** (single character _might_ be sufficient).
HW revision and boardloader version could be useful. A nice-to-have would be a capability structure, like:
* does boardloader support reboot-to-bootloader?
* do we have SD card?
* is the SD card hot-swappable?

We don't currently have an exact list of things to put there, besides the model identifier. At this point it is mostly a research topic, i.e., how to expose this data, how big it can be, etc.

prusnak, 2022-04-05:
Check `core/embed/prodtest/main.c` lines 384-391. 

That's where we set the id to OTP in production (only model and batch initial date is stored). There has been only 1 boardloader version deployed, so all devices have the same bootloader IIRC.

hiviah, 2022-04-11:
OTP seems readable before dropping privileged mode and setting MPU, however we will still need to put some data in boardloader to put the capabilities in. So the OTP could be used as fallback in case we don't have the new boardloader with the structure.

matejcik, 2022-06-14:
closed via #2324

closed 2022-06-14 by matejcik