trezor/trezor-firmware #2250: metamask message signature: error: firmware error

Kingway87, 2022-05-03:
trying to vote and stacking on evmos is not working , cosmo evm to erc

current firmware trezor : 2.4.3
metamask fireware 10.13/14

 - Wallet software: metamask
 - OS: Msc
 - Browser :Chrome, Firefox, brave
 

https://evmos.disperze.network/validators
after I click on stacking on validator and sign the message , the error pop up as metamask message signature: error: firmware error 

I contacted metamask support  and try all the method still cant solve it ,they ask me seek for trezor support..did send trezor support ticket but N Days never reply yet...may I know how ? a bit rush due to the claim rektdrop got expired date, this issue been more then 3 months.

Ophelia45, 2022-05-03:
I will suggest you contact metamask support and click on the live chat message logo to talk with a Developer from metamask through their live chat Developers forum to help you sort this issue out right away. https://metamaskzendesk.helpsconnect.live/#

matejcik, 2022-05-04:
@Kingway87 this is unfortunately not a support channel and we might need more technical information to be able to do anything. I will try to reproduce your issue though.

Please disregard a message you might have received from "Ophelia45", it was an obvious scam attempt.

matejcik, 2022-05-04:
Unfortunately I was unable to reproduce the issue because I don't have any EVMOS to stake.

This might be the same issue as #2036 or #2167. Please wait for the upcoming firmware release to see if the issue goes away. Otherwise please contact EVMOS support and let them give us the technical details we need. Feel free to refer to this issue.

For now, I am closing the issue. It will be reopened if relevant.

closed 2022-05-04 by matejcik

Kingway87, 2022-05-04:
okay noted thanks , if you need any evmos i can send you too ....yea did contact evmos team but no respond on this....