trezor/trezor-firmware #2253: KeyError: 27 Load key error in libcrypto

OpenGrid, 2022-05-04:
**Describe the bug**
After upgrading to Ubuntu 22.04 trezor-agent -c someuser@atsomehost
returns error 27

```
2022-05-05 00:45:01,636 WARNING      error: 27                                                                                            [server.py:100]
Traceback (most recent call last):
  File "/home/user/.local/lib/python3.10/site-packages/libagent/server.py", line 95, in handle_connection
    reply = handler.handle(msg=msg)
  File "/home/user/.local/lib/python3.10/site-packages/libagent/ssh/protocol.py", line 98, in handle
    log.warning('Unsupported command: %s (%d)', msg_name(code), code)
  File "/home/user/.local/lib/python3.10/site-packages/libagent/ssh/protocol.py", line 54, in msg_name
    return ids[code]
KeyError: 27
Load key "/tmp/trezor-ssh-pubkey-xxxxx": error in libcrypto
Permission denied (publickey).
```

**Firmware version and revision**
1.10.5

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: trezor-agent
 - Linux Ubuntu 22.04
 - python 3.10

**To Reproduce**
Use trezor-agent -c someuser@atsomehost

**Expected behavior**
connection with a previously configuration

**Additional context**
python and all pip packages are up to date

prusnak, 2022-05-05:
Please open an issue in the trezor-agent repository: https://github.com/romanz/trezor-agent

matejcik, 2022-05-05:
closing the issue as the traceback doesn't even involve Trezor

closed 2022-11-01 by hynek-jina