trezor/trezor-firmware #2254: UnexpectedMessage error when starting the device self test using trezorctl

ps1dr3x, 2022-05-05:
**Describe the bug**
UnexpectedMessage error when starting the device self testing via trezorctl.

**Firmware version and revision**
Bootloader version: 2.0.3
Firmware version: 2.4.3

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: trezorctl
 - OS: Linux, Ubuntu 20.04.4 LTS (fresh install)
 - Browser: N/A
 - Version: 0.13.0

**To Reproduce**
1. Connect the Trezor, boot it in bootloader mode
2. Run the command: trezorctl device self-test

**Expected behavior**
Self tests are executed correctly.

**Actual behavior**
"Error: UnexpectedMessage: Unexpected message".

**Additional context**
The Trezor is new, everything else seem to work fine, including the other commands from trezorctl.

prusnak, 2022-05-05:
Self-test is not meant to be run on a production firmware.

ps1dr3x, 2022-05-05:
I see, thanks. Being that the case I think the command should at least return a message explaining it though

ps1dr3x, 2022-05-05:
Just noticed also the command `trezorctl device load` is returning the same error message

matejcik, 2022-05-05:
these commands, along with `trezorctl debug *` and perhaps some others, should handle "UnexpectedMessageError" and print something sensible.

perhaps _all_ commands should handle `UnexpectedMessageError` and print something sensible?
 "This feature is not available on your Trezor" seems like a nice wording?

matejcik, 2022-05-05:
OTOH that might not be true -- in certain situations the feature is available but, e.g., the Trezor is in the wrong state to receive the message.

grdddj, 2022-05-05:
> Just noticed also the command `trezorctl device load` is returning the same error message, both in bootloader and normal mode.

`trezorctl device load` is also not supposed to be run on a "real hardware device", it needs a `debuglink`, which is present mostly in emulators for testing. In this case we should probably check if `debuglink` exists in `Trezor` and can report the problem correctly in negative case.

prusnak, 2022-05-05:
+1 for improving documentation
~0 for improving UnknownMessageException error handling
-1 for adding more code to handle these corner cases

trezorctl is meant to be used by people who know what they are doing

ps1dr3x, 2022-05-07:
> +1 for improving documentation ~0 for improving text when UnexpectedMessageError is encountered -1 for adding more complex code to handle these corner cases
> 
> trezorctl is meant to be used by people who know what they are doing

you need to know what you can do and how before you can know what you are doing 🙂

Anyway I agree with you, I didn't want to create a big hassle for the devs of this project, I just think it'd be nice to have a little explanation, even just in the help of the cli tool, to understand what you can and what you can't do

matejcik, 2022-05-09:
> 'Device does not support this function'

This is not necessarily true. In certain situations you can get "unexpected message" for something that is supported but unavailable at that moment.

My proposal would be to keep the part that says "Error: UnexpectedMessage: Unexpected message", and add to that something like "Are you sure your Trezor supports this feature?"

OTOH maybe the best solution is to modify/expand the help text for the affected functions.

Known examples are:
`trezorctl device self-test` -- works only on prodtest firmware
`trezorctl device load` -- works only on debug builds
`trezorctl debug *` -- work only on debug builds (but that is much more obvious being in the `debug` namespace
altcoin commands not supported on the other model, e.g., EOS or Monero operations on T1, CoSi signing on TT

bosomt, 2022-07-19:
@matejcik does this issue needs QA or i can approve it ?

btclinux, 2024-03-30:
> `trezorctl device self-test` -- works only on prodtest firmware

Can you explain more? Is prodtest firmware can run on production device or it's a something special? And, if yes, how to build prodtest firmware for model 1. Thank you!

matejcik, 2024-03-31:
It's a thing that runs -- or used to run -- in manufacturing. To test that the buttons click, the display paints, etc. You don't need it for anything, I'm not even sure _we_ need it for anything.
Support was removed in this commit: https://github.com/trezor/trezor-mcu/commit/66a8d1834855325e76a283b2d4fd76721213f9d6

commit c12e4edd (2022-06-30): fix(python/trezorctl): improve docs and behavior around unexpected messages

closes #2254