trezor/trezor-firmware #2295: Update Connect tests, account for its move to Suite monorepo

grdddj, 2022-05-26:
`Connect` tests are being moved into Suite monorepo.

We should account for this in our way of running them - https://github.com/trezor/trezor-firmware/blob/master/tests/connect_tests/connect_tests.sh

sime, 2022-05-26:
Could it (or should it?) potentially support both?

Since Connect v8 will be support for some time.

grdddj, 2022-05-27:
> Could it (or should it?) potentially support both?
> 
> Since Connect v8 will be support for some time.

Not sure at all. I am not knowledgeable enough about `Connect`, I just heard it moves to another place, so our `Connect` tests would no longer be updated

sime, 2022-05-27:
@mroz22 @szymonlesisz Thoughts on how to maintain Connect tests in firmware?

grdddj, 2022-09-26:
Already done, `Connect` tests are currently fine in `master`

closed 2022-09-26 by grdddj