trezor/trezor-firmware #2305: Broken build with python 3.8

TychoVrahe, 2022-05-27:
Latest changes (#2286)  broke build with python 3.8 and older due to some type annotations.

As python 3.8 is a default in some widely used OSs ( Ubuntu 20.04.4 LTS), it would make sense to support this version.

Nice to have would be to also test with the minimal python version in CI pipelines

prusnak, 2022-05-27:
Btw, reproducer is:

```
$ make gen
...
  File "...trezor-firmware/common/tools/coin_info.py", line 51, in <module>
    SupportInfo = dict[str, SupportInfoItem]
TypeError: 'type' object is not subscriptable
```

matejcik, 2022-05-27:
It seems easy enough to keep 3.7 or 3.8 compatibility. what i'm wondering about is how to ensure that breakages like this don't happen.

trezorlib runs tox over all supported Python versions so that is covered. the internal tooling, not at all. One way would be to set the nix env to use Python 3.8, but that is slightly complicated IIRC.
Maybe we have a linter available that would tell us this, and could run it as part of the `python test` CI job, where different Pythons are available?

PR #2308 (2022-05-30): Type hints fixes for older python versions
Fixes #2305:
- modifying type hints so that even older (`3.7` and `3.8`) python versions understand it
- also the test for this was created and added into `CI`
  - it is currently in the `Test` stage, but could be probably moved to `Prebuild` thanks to its nature (it is not dependent on `Build` and it is very similar to `style prebuild`)

In [this CI](https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/pipelines/549800707) the tests should fail, as it only has the first commit with tests - it is "fixed" in [subsequent CI](https://gitlab.com/satosh [... cut]
closed 2022-05-30 by grdddj