trezor/trezor-firmware #2310: Create new wallet (device setup) on Model R

sime, 2022-05-30:
**Design:** [Figma](https://www.figma.com/file/Fv3Ikqwm5eDXxnBoOQov4T/%5B1%23-Testing%5D-T1R?node-id=1402%3A4559)

|First screen|Second|
|-|-|
|[image]|[image]|

**User Story**
As a new Trezor user
I want setup my Trezor
So that I can hodl

**Acceptance criteria**
Given I connected my new Trezor
When I press 'Setup Trezor' in Suite
Then I should see 'Do you want to create a new wallet'
And I should see 'By continuning you agree to trezor.io/tos'
And I should see 'X' and 'Confirm' buttons

Given I see 'Do you want to create a new wallet'
When I press 'Confirm'
Then I should see 'Processing' animation
And I should see the homescreen with 'Needs Backup'

matejcik, 2022-06-13:
> It seems that the issue

what issue btw? :)

grdddj, 2022-06-13:
> > It seems that the issue
> 
> what issue btw? :)

Issue of not being able to create a new wallet via Suite - it gives `Firmware error` in `Suite` and that `icon error` in the emulator logs.

 Also locally with `trezorctl device setup`, it gives me just a black screen on emulator and a hanging `trezorctl`

grdddj, 2022-06-13:
> probably something to do with the icon being built conditionally -- if you build with `TREZOR_MODEL=R` and then run with `TREZOR_MODEL=R`, it will work; if you build with `TREZOR_MODEL=T` and run with `TREZOR_MODEL=R`, you might see the error. this naturally can't happen with `build_unix_frozen` because there is no build-time vs run-time mismatch

Just tried with a hardware device and also got the same error - both in `Suite` and in `trezorctl`:
```
247620000 trezor.wire ERROR exception:
Traceback (most recent call last):
  File "trezor/wire/__init__.py", line 341, in _handle_single_message
  File "trezor/loop.py", line 533, in __iter__
  File "trezor/loop.py", line 184, in _step
  File "apps/management/reset_device/__init__.py", line 37, in reset_device
  File "trezor/ui/__init__.py", line 372, in __iter__
  File "trezor/loop.py", line 352, in __iter__
  File "trezor/loop.py", line 186, in _step
  File "trezor/ui/__init__.py", line 458, in handle_rendering
  File "trezor/ui/loader.py", line 134, in dispatch
  File "trezor/ui/__init__.py", line 288, in dispatch
  File "trezor/ui/loader.py", line 108, in on_render
ValueError: Invalid icon size
```