trezor/trezor-firmware #2318: Error screen for PIN creation failure

sime, 2022-06-01:
**Design:** 
[image]

Similar to wrong pin when attempting an unlock.

[image]

**Spec:** 
When establishing a PIN, a mismatched PIN will prompt the user about the PIN entry failure. Cancelling will return the user to the homescreen.

grdddj, 2022-06-13:
In [b476fc7](https://github.com/trezor/trezor-firmware/pull/2277/commits/b476fc702381f36de0cd1372d70a9f41053b5228) I have added the "show popup" functionality for `model R`, matching the current `model T` popup logic.

It is just a popup that will be on the screen for 3 seconds and then disappear, leaving the start of the PIN dialogue there.

> Cancelling will return the user to the homescreen.

Do we really want this? The behavior of the `model T` is as described above. Users can cancel during the PIN creation

sime, 2022-06-13:
> > Cancelling will return the user to the homescreen.
> 
> Do we really want this? The behavior of the `model T` is as described above. Users can cancel during the PIN creation

Can't fault the existing cancel behaviour. You can ignore this.

closed 2022-06-21 by sime