trezor/trezor-firmware #2330: Model R: Homescreen

sime, 2022-06-08:
## Design
[Figma](https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=1830%3A10277)

[image]

### Default home screen
128x64:
[image]

### Font
[Pixel Operator 8](https://github.com/trezor/trezor-firmware/issues/2269)

## User Story
As a Trezor user
I want to see the homescreen
So that I know my Trezor is ready to use.

## Acceptance Criteria
Given my Trezor has a PIN set
and the auto lock time passes
Then the screen should dim
And I see LOCKED on the screen

Given my Trezor is LOCKED
When I tap any button
Then I should be prompted for my pin
And the screen brightness is at 100%

grdddj, 2022-07-24:
Done in [1df6093](https://github.com/trezor/trezor-firmware/pull/2277/commits/1df60935e9965cea5192f537f917916344ddd781)

sime, 2022-07-25:
**QA NOK**

* Missing the [default homescreen](https://user-images.githubusercontent.com/216917/172592735-3efde758-5f47-4c3a-948b-e490eb42d826.png).
* Distance between lock icon and text should be 4px
* Device label when not connected, should be 2px from the top
* 'Click to connect' should be 13px from the bottom (or 8px from 'Not Connected')

Screenshot for reference
[image]

grdddj, 2022-07-25:
> **QA NOK**
>
>   Missing the [default homescreen](https://user-images.githubusercontent.com/216917/172592735-3efde758-5f47-4c3a-948b-e490eb42d826.png).
>
>    Distance between lock icon and text should be [...]

All should be addressed in [e9f2f7f](https://github.com/trezor/trezor-firmware/pull/2277/commits/e9f2f7f09443c7d9d9651c07812b9692f2e4ab49)

sime, 2022-07-28:
**QA NOK**

@grdddj 

> Distance between lock icon and text should be 4px
Still not the case

`NOT CONNECTED` should have a `2px` padding between the letters, currently it is 1px.

`LOCKED` screen should show device label at the bottom, and nothing on the top.

sime, 2022-07-28:
Reference image

[image]

grdddj, 2022-07-28:
> LOCKED screen should show device label at the bottom, and nothing on the top.

I saw it in Figma, but it is missing the `Click to unlock` thing there. Should all three things (locked, unlock and label) be put on the bottom?

sime, 2022-07-28:
> > LOCKED screen should show device label at the bottom, and nothing on the top.
> 
> I saw it in Figma, but it is missing the `Click to unlock` thing there. Should all three things (locked, unlock and label) be put on the bottom?

Ah you are right. Leave the content of the `LOCKED` screen for now.

closed 2022-08-04 by sime