trezor/trezor-firmware #2331: Model R: Banners

sime, 2022-06-08:
## Design
[Figma](https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=1830%3A11937)

### Font
[Pixel Operator 8](https://github.com/trezor/trezor-firmware/issues/2269)
 
## User Story
Given I setup my Trezor
When I skip the backup
Then I should see 'BACKUP NEEDED'

grdddj, 2022-07-24:
Done in [1df6093](https://github.com/trezor/trezor-firmware/pull/2277/commits/1df60935e9965cea5192f537f917916344ddd781) - adding the warning icons in top corners and changing the text to monospace white

sime, 2022-07-25:
@grdddj Get ready for some pixel pushing.

* Banner is 4px from the top, it needs to be **2px**.
* Top right icon is clipped by one pixel, it's showing 10x11 when the icon is actually **11x11**.
* Banner text is not centred.
* Word spacing needs to be **9px**.

Current screenshot for reference.

[image]

grdddj, 2022-07-25:
In [2ffed0b](https://github.com/trezor/trezor-firmware/pull/2277/commits/2ffed0b22ff7566996a4da52c5214e730154eb83):
- moving banner `2px` to the top
- moving the right icon `1px` to the left
- helping the centering text by `removing !` from the text (users in survey were not happy about it and warning icons already do the job)

About the word spacing, how many it is right now? I could look into how we render a space

[image]

grdddj, 2022-07-25:
> Word spacing needs to be 9px.

It looks like the `Unifont` has it **8px** and `Mono` has **10px**

sime, 2022-07-27:
> * moving the right icon `1px` to the left

I'm using this tool for [image comparison](https://online-image-comparison.com/), and the right icon looks like it shift 2px to the left.
[image]

Reference image

[image]

grdddj, 2022-07-28:
> It looks like the `Unifont` has it **8px** and `Mono` has **10px**

Unified to **9px** for both fonts in [a9627ab](https://github.com/trezor/trezor-firmware/pull/2277/commits/a9627ab266d9b0f6c1cb2496ed8f60dc2a69f37b)

grdddj, 2022-07-28:
> > * moving the right icon `1px` to the left
> 
> I'm using this tool for [image comparison](https://online-image-comparison.com/), and the right icon looks like it shift 2px to the left.

This is because of a known limitation of our icon creation/rendering. I have an issue for that - https://github.com/trezor/trezor-firmware/issues/2416.

It is not a show-stopper, we can make an easy workaround for it - just creating a second version of the icon that will be rendered in the right top. However, it would be nice to address the limitations.

BTW, nice comparison tool, thanks for sharing :)

sime, 2022-07-28:
> This is because of a known limitation of our icon creation/rendering. I have an issue for that - #2416.
>
> It is not a show-stopper, we can make an easy workaround for it - just creating a second [...]

Would it work if the banner icon can be globally 12x11?

grdddj, 2022-07-28:
> Would it work if the banner icon can be globally 12x11?

If its width would be `12`, then yes, it would help us. But how we could add one pixel to the width in this symetric shape?

sime, 2022-07-28:
From a raster format POV it's trivial. Pad the image with a column of empty pixels.
[image]

grdddj, 2022-07-28:
> From a raster format POV it's trivial. Pad the image with a column of empty pixels

I actually need to do it already for this icon, as we cannot work with `11x11`, so I have it `12x12` with empty bottom row and right column.

The issue is that this is suitable for the `left corner` - in the `right corner`, we would need empty left column to show it properly "cornered"

grdddj, 2022-08-02:
Special icon for right corner created and used in [8bf08d9](https://github.com/trezor/trezor-firmware/pull/2277/commits/8bf08d94ff9d6bab693ce0359eb5a8c04f9adbbc)

closed 2022-08-04 by sime