trezor/trezor-firmware #2338: ed25519 unsafe signing

kchalkias, 2022-06-16:
A recent security report revealed a potential private key leak issue when the `ed25519.sign` api takes a public key as input, see [MystenLabs/ed25519-unsafe-libs](https://github.com/MystenLabs/ed25519-unsafe-libs)

TL;DR Most projects with a dependency to donna library tend to allow this public api for their devs and an accidental misuse (i.e. provide non matching pub-priv keys as signing function inputs) can be exploited by leaking the private key.

Trezor's implementation relies on the ed25519-donna api, which can be accidentally misused by Trezor devs, see here: https://github.com/trezor/trezor-firmware/blob/master/crypto/ed25519-donna/ed25519.c#L110

It is highly recommended that the sign api should not take the public key as input and it should be either 

- computed internally to the signing function (as the ed25519 RFC 8032 recommends) 
or 
- refactor your private key structure to include the public key bytes too - many projects take this approach with the aim to avoid recomputing the pubKey on each sig. request, i.e., similar to https://github.com/penumbra-zone/ed25519-consensus/blob/main/src/signing_key.rs#L16

prusnak, 2022-06-16:
There are only 2 places where we call `ed25519_sign`:

1. https://github.com/trezor/trezor-firmware/blob/7d37109eb84ef029e7b7ef3eac55db3237af03f6/core/embed/extmod/modtrezorcrypto/modtrezorcrypto-ed25519.h#L97-L98
2. https://github.com/trezor/trezor-firmware/blob/7d37109eb84ef029e7b7ef3eac55db3237af03f6/legacy/firmware/stellar.c#L1371

Case 1 calls `ed25519_publickey` to compute the public key from the secret key just beforehands
Case 2 uses the structure `node` which stores the corresponding public key to the private key.

So far I can tell the Trezor code is not affected.

kchalkias, 2022-06-16:
This is true and already cross checked that, there is no issue with current use, but the problem is by exposing this api there might a future one

andrewkozlik, 2022-06-16:
There are two more places in https://github.com/trezor/trezor-firmware/blob/master/core/embed/extmod/modtrezorcrypto/modtrezorcrypto-ed25519.h, where similar functions `ed25519_sign_keccak()` and `ed25519_sign_ext()` are used, but those are not affected either, because the public key is computed immediately before calling the function. Since we almost always compute the public key before calling sing, we might as well do it inside the sign function. It seems like an unnecessary and premature optimization to save a few clock cycles in T1 Stellar signing. So I wouldn't mind removing the `pk` from the ed25519 signing function's API.

andrewkozlik, 2022-06-23:
I will start work on removing `pk` from the ed25519 signing function's API.

closed 2022-06-27 by andrewkozlik