trezor/trezor-firmware #2343: Model R: Add hold to confirm to device wipe

sime, 2022-06-17:
**Design:** [Figma](https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=1830%3A10214)

**Spec:**
Given I am a Trezor user
When I click Factory reset in Trezor Suite
Then I should see 'Hold to Confirm'

Given I am on wipe device screen on Trezor
When I press hold to confirm for 2 seconds
Then my device should be wiped
And should be rebooted

grdddj, 2022-07-11:
Done in https://github.com/trezor/trezor-firmware/pull/2277/commits/4b694273016171df3e7ffe60a9cbb2e48d0569b4 as a part of https://github.com/trezor/trezor-firmware/pull/2277

sime, 2022-07-15:
**QA NOK**

* Upon completion of the animation ('loaded state'), the button is goes back to initial ('unloaded') state. The animation should remain loaded.
* Noticeable delay between the animation is complete and the transition to the next screen.

https://user-images.githubusercontent.com/216917/179198558-1ae6ac54-6ba5-4a5b-b6d7-5f0aa5eec997.mp4

grdddj, 2022-07-15:
> * Upon completion of the animation ('loaded state'), the button is goes back to initial ('unloaded') state. The animation should remain loaded.

Improved in [606127d](https://github.com/trezor/trezor-firmware/pull/2277/commits/606127da508f6b417b0a3317e15bd627e411f26c) - the hold-to-confirm stays in the filled/loaded state at the end

sime, 2022-07-15:
*QA OK*

closed 2022-07-15 by sime