trezor/trezor-firmware #2350: Deprecate EthereumGetPublicKey

matejcik, 2022-06-24:
The functionality is identical to `GetPublicKey`, because Ethereum "xpubs" are actually encoded exactly the same way as Bitcoin's. 

We can get rid of `get_public_key.py` in Ethereum by redirecting the handler to `bitcoin/get_public_key.py`, which should Just Work(tm) until someone adds a required field to `GetPublicKey` that isn't sent by `EthereumGetPublicKey` callers.

For this reason, we should _also_ deprecate `EthereumGetPublicKey` message and tell users to not use it anymore.
cc @mroz22 for Connect impact.

prusnak, 2022-06-24:
Each idea should start with a motivation. What's the benefit of doing this? I don't see any.

matejcik, 2022-06-24:
identical functionality, less code to maintain

matejcik, 2022-06-24:
one snag is that `EthereumGetPublicKey` has its own `EthereumPublicKey` return type, which is also identical to bitcoin's `PublicKey`.

this means that internally we need to keep the handler, but invoke bitcoin's handler to do the actual work

--- 

Currently `GetPublicKey` has no path checks -- when we tried to add them last time, we ran headlong into people actually using `GetPublicKey` with completely arbitrary paths, and/or for non-Bitcoin currencies.

This makes some amount of sense, and I'm not sure if there is a reasonable way to add path checks to the call.
That said, if in the future we manage to add path checks, it will have been useful to keep `EthereumGetPublicKey` separate, because it indicates different set of paths to allow.
_That_ said, Bitcoin's `GetPublicKey` allows one to specify coin name, whereas Ethereum's does not. We technically need to allow all supported coins / networks for one of the calls anyway.

prusnak, 2022-06-24:
I think this small improvement doesn't justify the overhead we'll created literally everywhere. 

If the code is bloated we can refactor it, but lets keep both messages please.

matejcik, 2022-06-24:
Eh, you're right, it's not worth the hassle.

closed 2022-06-24 by matejcik