trezor/trezor-firmware #2351: Prepare a system to embed bootloader version according to model type in core

TychoVrahe, 2022-06-27:
Currently we only have a bootloader binary for model T in Core source. We will need to design a way to do it for additional models. Obviously a solution like bootloader_T.bin and bootloader_R.bin is possible, but maybe there might be a better way to do it? something that will not bloat the repo over time with binaries, like pulling the bootloader binary from CI during build or something?

prusnak, 2022-06-27:
> bootloader_T.bin and bootloader_R.bin is possible

+1

The binaries are really small, there is no need to over-optimize right now.

matejcik, 2022-12-19:
given that we have predefined internal names, the solution is now clear:
`bootloader_T2T1.bin`
`bootloader_T2B1.bin`

PR #2731 (2023-01-20): Differentiate build results for different models
Adds a named binary file with added information about variant, model and version. Original files are left available too.

the filename is structured this way:
`<binary type>-<internal model>-<variant, if any>-<version>.bin`
i.e.
`firmware-T2T1-btconly-2.5.4.bin`
`bootloader-T2T1-2.0.5.bin`

Also a system enabling different embedded bootloaders according to model is implemented.

closes #2351 
closes #2697
closed 2023-01-20 by TychoVrahe