trezor/trezor-firmware #2356: MacOS: auto-patchelf-hook is not supported on ‘aarch64-darwin’

matejcik, 2022-06-30:
`nix-shell` refuses to start for me because of auto-patchelf-hook.

the obvious solution is to move it to the non-Darwin section, the only question is whether that will cause any sort of problem.
presumably not because it's only used to patch emulators that won't work on Darwin anyway?

matejcik, 2022-06-30:
workaround for me is exporting `export NIXPKGS_ALLOW_UNSUPPORTED_SYSTEM=1` which nix-shell helpfully suggests. but it's not nice that this is the default behavior

prusnak, 2022-06-30:
Oddly enough, nix-shell works for me on aarch64-darwin. That said I am pretty sure we can move `autoPatchelfHook` to `lib.optionals (!stdenv.isDarwin)` section of `shell.nix`

prusnak, 2022-06-30:
FTR I just tested `release-22.05` branch of nixpkgs and see the same error:

```
error: Package ‘auto-patchelf-hook’ in /nix/store/hcsl8zxvg8jnncid40xnm3drlvdcqd4c-source/pkgs/build-support/trivial-builders.nix:74 is not supported on ‘aarch64-darwin’, refusing to evaluate.
```

PR #2377 (2022-07-09): chore(nix): don't install autoPatchelfHook on MacOS
Fixes #2356.
closed 2022-07-09 by mmilata