trezor/trezor-firmware #2358: Model R: Carousel PIN entry

sime, 2022-06-30:
**Design:**
[image]

[image]

**Spec:**
Given I am on PIN entry
And I see 'Enter new PIN' in the center
When I press left 4 times
Then I should see the number 9 in the center

Given I am on PIN entry
And I have entered a PIN
And I have 'Show PIN' in the center
When I tap 'Confirm'
Then I should see the PIN in the top

Given I am on PIN entry
And I have entered a PIN
And I have 'Delete' in the center
When I tap 'Confirm'
Then I should delete one digit of my PIN

Given I am on PIN entry
And I have 'Exit' in the center
When I tap 'Confirm'
Then I should be on the homescreen

sime, 2022-07-14:
From the QA POV.

The hold to confirm action when PIN digits confirming by dual pressing needs to be removed.

grdddj, 2022-07-22:
> The hold to confirm action when PIN digits confirming by dual pressing needs to be removed.

Removed in [0b3589f](https://github.com/trezor/trezor-firmware/pull/2277/commits/0b3589fb2f2519dbb07d7f16b57f3cd78af26c82)

Carousel for `passphrase` added in [d36fcc8](https://github.com/trezor/trezor-firmware/pull/2277/commits/d36fcc8720834c383114a40391f74719de21caf4).

closed 2022-09-26 by grdddj