trezor/trezor-firmware #2387: [Cardano] Invalid Witness Request during signing with Withdrawal using non-account#1 staking path

burtbot25, 2022-07-14:
**Describe the bug**
I'm having problems signing with Trezor Connect when trying to sign a withdrawal transaction, but only when I'm using a staking path that is not the first account.

For example:

When signing a Cardano withdrawal transaction with **m/1852'/1815'/0'/2/0** in the withdrawal parameters, signing is successful.

However, signing fails when signing a withdrawal made on a staking path of a different account such as **m/1852'/1815'/1'/2/0** or **m/1852'/1815'/2'/2/0**. In these cases I get an "**Invalid witness request**" error returned from trezor and I am unable to get a signature.

**Firmware version and revision**
2.5.1

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Connect
 - OS: Windows
 - Browser: Chrome
 - Version: 104.0

**To Reproduce**
1. Create an ordinary transaction which include Withdrawal parameters
e.g.,: "withdrawals":[{"path":"m/1852'/1815'/1'/2/0","amount":"2182345"}]
2. Using web Trezor connect, confirm signing
3. Confirm signing on Trezor (Model T)
4. After "Hold to confirm", "Invalid witness request" error appears in the browser console

**Expected behavior**
Successfully sign the transaction and return a signature

matejcik, 2022-07-15:
cc @davidmisiak or @gabrielKerekes

gabrielKerekes, 2022-07-18:
This is a safety feature and expected behaviour. We don't allow using multiple accounts in a single transaction. The issue was with the possibility of an attacker sending the user's funds to different accounts (owned by the user). Outputs containing an address belonging to any of the user's accounts were considered change outputs and wouldn't be displayed to the user which would allow an attacker to deanonymize the user's accounts without the user even knowing about it.

While implementing this feature we also decided to extend the "policy" to all the transaction elements. And so you can not make a cross-account withdrawal. I admit this might be a bit of an overkill for withdrawals as their path was always being shown, but this won't be true anymore with the introduction of "expert mode" (https://github.com/trezor/trezor-firmware/pull/2355) which should be included in the next release.

There's extensive conversation on the topic in https://github.com/trezor/trezor-firmware/pull/1683#discussion_r689517936.

matejcik, 2022-07-18:
thanks, I was not sure how withdrawals plug into this.
closing as intended behavior.

closed 2022-11-01 by hynek-jina