trezor/trezor-firmware #2389: Model R: Introduce Cancel/'X' icon

sime, 2022-07-15:
Design: [Figma](https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=1830%3A10214)

[image]

Example location: Hold to wipe

grdddj, 2022-07-23:
Support for icons as buttons added in [cfc9e2e](https://github.com/trezor/trezor-firmware/pull/2277/commits/cfc9e2e68f4ba38c5fd67fec0d7f383d5f5bea9b)

It currently includes these:
- small 8*8 icon with an outline/border ("x")
- bigger icon up to 12*12 without an outline ("<", ">")
- "arms" on the left and right of the middle button
- arrow-buttons with customizable width for scrolling the content up and down

sime, 2022-07-27:
Borderless `X` icon is needed.

No actual screens are using a 'bordered' `X` icon.

sime, 2022-07-28:
Additionally, remove borders around these buttons, and select needs arms.

[image][image]

grdddj, 2022-08-02:
Done in [c163b03](https://github.com/trezor/trezor-firmware/pull/2277/commits/c163b033eea86620b4683248eb0a34db012109d8) and [b185fd0](https://github.com/trezor/trezor-firmware/pull/2277/commits/b185fd0f296ca9d1480b4ee4c09453cf3880b6da)

closed 2022-08-04 by sime