trezor/trezor-firmware #2394: Thousand amount separator

hynek-jina, 2022-07-20:
With the use of `sat` units, there was a need to implement thousand separators for amount in Trezor Suite.
https://github.com/trezor/trezor-suite/issues/5545

The user needs to see the amount in the Suite and on the Trezor device in the same format for control when sending a transaction.

Most required is `en-us` formating (like ie `123,456.78`) but more general solution would be even better for the future.

sime, 2022-07-21:
Having a hardcoded locale scheme for a thousand separator when sats are used should be trivial.

We want to avoid locale's that [support spaces](https://docs.oracle.com/cd/E19455-01/806-0169/overview-9/index.html). Then a single scheme can cover the majority of cases.

@matejcik We would like to avoid spaces for security POV?

hynek-jina, 2022-07-21:
Then we can go simply with `,` thousand separator which is currently in Suite as well.

closed 2022-08-02 by matejcik

bosomt, 2022-08-05:
**QA OK**

TT OK

**Info:**
 - Suite version: desktop 22.8.1 (503b472935dec8a9c07aac5eef5ba43f86780a38)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.8.1 Chrome/100.0.4896.160 Electron/18.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1680x1050
 - Device: model T 2.5.2 Universal (revision 0d87b55ba4fed7eecc72bf2a94ee473830b095e9)
 - Transport: bridge 2.0.31

bosomt, 2022-08-05:
Maybe we can consider to adding this separator to Trezor Suite too ?
It would look clearer.

[image]

hynek-jina, 2022-08-05:
@bosomt 
It's almost there: https://github.com/trezor/trezor-suite/issues/5545

commit 563f6ec3 (2022-07-29): feat(core/strings): use thousands separator (fixes #2394)
commit 2d1b60b2 (2022-08-01): feat(legacy): use thousands separator (fixes #2394)
commit eff99099 (2022-08-01): feat(core/strings): use thousands separator (fixes #2394)
commit f58b636d (2022-08-01): fixup! feat(legacy): use thousands separator (fixes #2394)