trezor/trezor-firmware #2399: Model R: bigger fonts for PIN and other input methods

grdddj, 2022-07-25:
The middle character in choice should be `64` font size, the side ones `32`.

It might eat a lot of flash space, export just the necessary characters, or try to somehow reuse already existing fonts (sizes `8` and `16` - maybe just "multiplying" them by four)

matejcik, 2022-07-25:
> somehow reuse already existing fonts (sizes `8` and `16` - maybe just "multiplying" them by four)

this is the way to go.

grdddj, 2022-07-28:
Magnification supported in [a4b8248](https://github.com/trezor/trezor-firmware/pull/2277/commits/a4b82488f9833fee26af02ba70b76383c7fa8c8e) and used in [85fc691](https://github.com/trezor/trezor-firmware/pull/2277/commits/85fc6910b3a9b50a69dc18eaab2c6f43de7ff668)

I am very interested in how much flash space has the transformation logic eaten, I have not yet looked.

Couple letters still have some small inconsistencies, for example
[image]
[image]

grdddj, 2022-07-28:
> I am very interested in how much flash space has the transformation logic eaten, I have not yet looked.

Seems it did not make a big difference, around `2kb`, which is negligible compared to including all those big fonts.

grdddj, 2022-07-28:
As agreed with @matejcik, the magnification could be put directly into already existing rendering logic

closed 2022-08-25 by hynek-jina