trezor/trezor-firmware #2407: Model R: Bitcoin signing flow design

grdddj, 2022-07-25:
Implement all the BTC signing screens according to `Figma` design.

grdddj, 2022-08-02:
The first part is roughly done in [ae5960e](https://github.com/trezor/trezor-firmware/pull/2277/commits/ae5960e7858337a48b23d10b5b2f5e1991ac2a8b) - without much pixel-play and some details (for example the down arrow next the address is missing)

[image]
[image]

I am omitting the middle screen completely, as it does not make sense to show the fee and total (plus the fee-rate possibly) for each output - that will be part of the final transaction confirmation - to be done.

Currently pressing `X` from the summary goes to the previous page, we want to probably make it cancel the whole signing flow (that can be currently done by pressing `X` at the first screen).

Other considerations:
- Figma shows this flow in different versions, so some final design would need to be made
- We might include the horizontal scrollbar at the top (as is in some Figma designs) and/or consistent title on all screens (like `SEND`)

To be done:
- final transaction screen showing the total amount, fee and fee-rate
  - it can theoretically look something like the middle screen above, but with extra space for fee-rate

grdddj, 2022-08-04:
> Tried to run the UI on commit https://github.com/trezor/trezor-firmware/commit/ec8dd5800b4b5589efaed2282a3ff2bb830c05f7
It crashed the emulator.

Good catch. The issue was the length of the address - I allocated just 50 characters on it, and `taproot` addresses have 62 characters :) 

Fixed in [3e4d83d](https://github.com/trezor/trezor-firmware/pull/2277/commits/3e4d83d8e4827d18e06c40d0c649dd4b2f908ab2) - allocating 80 characters.

However, for `taproot` address, the last two characters are not very nice right now, colliding with the left button. Something will need to be done with it - some truncating, splitting it into two pages or displaying the characters more densely in the middle rows, so that more fit in 5 rows.

[image]

sime, 2022-08-04:
We will need multiple pages for sure. As other coins have longer addresses.

grdddj, 2022-08-07:
> We will need multiple pages for sure. As other coins have longer addresses.

Paginated flow pages supported in [28071d8](https://github.com/trezor/trezor-firmware/pull/2277/commits/28071d8252e5541fba845aa9544af3c1c399be4a)

[image]
[image]

closed 2022-11-10 by hynek-jina