trezor/trezor-firmware #2416: Model R: investigate the limitations of icons rendering

grdddj, 2022-07-28:
Currently, there are a couple of limitations regarding icons/images:
- `TOIF` format does not work with odd-width icons, so we need to leave one column blank when the icon is odd-wide
- we cannot start drawing icons at an odd `x` coordinate, which means the icon needs to be also modified for this
- it seemed that when the height was odd, it refused to draw the last line - so all icons need to be even even-wide

It would be nice to get an idea of how difficult it would be to fix these, as it would make the precise-pixel-work easier.

prusnak, 2022-12-16:
Crazy idea:

How about we treat the display as it were 120x120 pixels, but with double resolution subpixels (iOS marks this as 120x120@2)?

In another word, we should never allow an element to be at odd coordinates and/or an element to be odd width/height.

Since we now have UI in Rust, would it be possible to throw a compile-time error when we try to render an element at odd coordinates and/or an element with odd width/height?

My rationale is: if you need to render an element at odd coordinates and/or with odd dimensions, there is something wrong with how the element is designed.

matejcik, 2022-12-16:
This breaks down on TR's 128x64 display though :(

prusnak, 2022-12-16:
> This breaks down on TR's 128x64 display though :(

Yeah, I am talking about hi-res displays, such as the Model T display.

TT: 120x120@2
TR: 128x64@1

prusnak, 2022-12-16:
If this issue is mostly about TR, then let's ignore my comments.

TychoVrahe, 2022-12-16:
Why should having odd dimensions be considered wrong though? Design does this occasionally for some icons.

Even more with placement, when icon is 10x10px but you want it centered on the screen this transalates to x-coordinate= 115, so we would further need to limit on dimensions being divisible by 4 to enable centering when placement on odd coords is prohibited.

Or lets say some animations, e.q. marquee - we would need to skip every other frame potentially affecting smoothness

prusnak, 2022-12-16:
Fair enough.

matejcik, 2023-02-02:
some parts of this have been done in #2512 , others we will leave like it is now

closed 2023-02-02 by matejcik