trezor/trezor-firmware #2417: Alter distribution of code into .flash and .flash2 sections

grdddj, 2022-07-28:
Currently, with both the `TT` redesign and `model R` development, we are transferring `UI` code (and potentially other things) from `micropython` into `Rust`.

`Rust` code resides in `.flash` and `micropython` code in `.flash2` regions/sections. As we make the transition, `.flash` is overflowing while `.flash2` has (will have) a lot of free space (even more after implementing https://github.com/trezor/trezor-firmware/issues/15).

We might want/need to take some stuff from `.flash` and put them into `.flash2` to level it out (or most importantly avoid `.flash` overflowing).

mmilata, 2022-09-09:
Please note that `.flash` is in in `rx` segment while `.flash2` is in `r`, so we can't move executable code there.

I think there's a gap between these two segments so we can't adjust the sizes by moving a boundary. However it seems to me that we could swap them entirely:
* `.flash` 768K -> 896K (+128K)
* `.flash2` 896K -> 768K (-128K)

hiviah, 2022-09-12:
We can change MPU to have `.flash2` set to `rx`.

TychoVrahe, 2022-09-29:
seems to me that both .flash and .flash2 are already executable or am i missing something?

hiviah, 2022-09-29:
You're right, .flash2 doesn't have NX flag.

mmilata, 2022-09-30:
Ah, so the beginning of memory_T.ld does not actually correspond to the memory controller configuration?
```
MEMORY {
  FLASH  (rx)  : ORIGIN = 0x08040000, LENGTH = 768K
  FLASH2 (r)   : ORIGIN = 0x08120000, LENGTH = 896K
  CCMRAM (wal) : ORIGIN = 0x10000000, LENGTH = 64K
  SRAM   (wal) : ORIGIN = 0x20000000, LENGTH = 192K
}
```

TychoVrahe, 2022-09-30:
found some explanation here: http://web.mit.edu/rhel-doc/3/rhel-ld-en-3/memory.html

tldr: this instructs the linker to place sections which are not explicitly mapped into regions into the regions defined here according to the attributes.

since we are mapping all the sections explicitly later in `SECTION`, with `>FLASH` etc, this should not have impact.

once the binary is loaded to the device, executability is controlled by MPU, where it is allowed for both regions.

we could define the `FLASH2 `as `(rx)` to make it clearer though.

PR #2534 (2022-10-03): Redistribute contents in flash 
This PR redistributes content in flash a bit, so that all UI1 and UI2 on TT and TR can be linked. TT with UI2 and PYOPT=0 however does not fit, there just isn't enough space left in both sections to fit that right now. 

With logic that we are moving UI from python to rust, i moved the rust library into same section as python, so that overall size of that section is not changed too much when code is rewritten from py to rs.

closes #2417
closed 2022-10-03 by TychoVrahe