trezor/trezor-firmware #2438: Fractional fee rates are incorrectly displayed with latest develop version

bosomt, 2022-08-05:
**Describe the bug**
Fractional fee rates are incorrectly displayed with latest develop versions.
I was not able to reproduce issue on production versions of firmwares /only final fee is displayed and not fee rate/

**Info:**
 - Suite version: desktop 22.7.3 (8eddb2accd31c78577f276df2e9ecf773d2c3c17)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.7.3 Chrome/100.0.4896.160 Electron/18.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1680x1050
 - Device: model T 2.5.2 regular (revision 426eae4dfcb853041419030bbb104cc2bbf6629f)
 - Transport: bridge 2.0.31

**To Reproduce**
Steps to reproduce the behavior:
1. Navigate to Send dialogue
2. Enter custom fee with fractions
3. Compare fee rate in Suite and on device during TX signing

**Screenshots**
[image]

matejcik, 2022-08-05:
and resolved in #2431

bosomt, 2022-08-05:
**QA NOK**

sorry :(

[image]

**Info:**
 - Suite version: desktop 22.8.1 (503b472935dec8a9c07aac5eef5ba43f86780a38)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.8.1 Chrome/100.0.4896.160 Electron/18.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1680x1050
 - Device: model T 2.5.3 Universal (revision ef47c262720be8b21671e85d7af68c547ad1955e)
 - Transport: bridge 2.0.31

matejcik, 2022-08-05:
well that's a big difference

can you give us the tx parameters? or the raw signed tx?

matejcik, 2022-08-05:
tx inputs:
```
[
  {
    "address_n": [
      2147483732,
      2147483649,
      2147483648,
      0,
      0
    ],
    "prev_index": 0,
    "prev_hash": "a55a49138d2dbf49f344e3fd3fd7821593c01daec1ab340a845ea328dac2cdd1",
    "script_type": "SPENDWITNESS",
    "amount": "1198145",
    "sequence": 4294967293
  }
]
```
outputs:
```
[
  {
    "address": "tb1q59gtxd5g4zzlu4l4u5dg4z00j2qxxsmu9tw9d6",
    "amount": "1000",
    "script_type": "PAYTOADDRESS"
  }
]
```
raw signed tx (invalid signature):
```
02000000000101d1cdc2da28a35e840a34abc1ae1dc0931582d73ffde344f349bf2d8d13495aa50000000000fdffffff01e803000000000000160014a150b33688a885fe57f5e51a8a89ef928063437c02473044022060b7cf4bf51fcc9697f08464ecb143688aadf707bf2338baea0455bb3860bf4502207245839d19ae19bad504ce5c0349dd5868133e3ae8c6937dccea1567c963b477012103adc58245cf28406af0ef5cc24b8afba7f1be6c72f279b642d85c48798685f86200000000
```

an independent library counts vsize as `110` which matches Trezor's fee rate but not Suite's. @mroz22 can you take a look?

bosomt, 2022-08-08:
**QA OK**

tested on latest dev master with 2.5.2 FW

[image]

**Info:**
 - Suite version: web 22.9.0 (3fbbabb1b10ea6ebebd2ecf926cb8dc412d9cc16)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:103.0) Gecko/20100101 Firefox/103.0
 - OS: MacIntel
 - Screen: 1680x1050
 - Device: model T 2.5.2 Universal (revision 0d87b55ba4fed7eecc72bf2a94ee473830b095e9)
 - Transport: bridge 2.0.31

mroz22, 2022-08-08:
> **QA NOK**
> 
> sorry :(
> 
> [image]
> 
> **Info:**
> 
> * Suite version: desktop 22.8.1 (503b472935dec8a9c07aac5eef5ba43f86780a38)
> * Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/22.8.1 Chrome/100.0.4896.160 Electron/18.3.5 Safari/537.36
> * OS: MacIntel
> * Screen: 1680x1050
> * Device: model T 2.5.3 Universal (revision [ef47c26](https://github.com/trezor/trezor-firmware/commit/ef47c262720be8b21671e85d7af68c547ad1955e))
> * Transport: bridge 2.0.31

This could be related to this https://github.com/trezor/trezor-suite/issues/5907
@sime

mroz22, 2022-08-08:
`110` in suite as well. 

In some cases, it could be this issue https://github.com/trezor/trezor-suite/issues/5907

but in this case it does not feel like that because if you have vsize `110` and feeRate `8888.99` it makes up `977788.9` so there is plenty of space to create change output

bosomt, 2022-12-19:
**QA OK**

fixed 🎉  , tested  at least 10 fee rates up to 8888.88 or 9999.99 and it always matched /device vs Suite/

**Info:**
 - Suite version: desktop 23.1.0 (38b40ef729793e945c3ad2c1f45e29db49460311)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/23.1.0 Chrome/104.0.5112.124 Electron/20.3.5 Safari/537.36
 - OS: MacIntel
 - Screen: 1440x900
 - Device: model T 2.5.4 Universal (revision 305b7fe84d802db7f0826d07a222a886f73732f0)
 - Transport: bridge 2.0.32

closed 2022-12-19 by bosomt