trezor/trezor-firmware #2440: Model R: unite the Rust API with redesigned model T

grdddj, 2022-08-06:
It would be nice if the API between `micropython` and `Rust` was completely the same for `model T` and `model R`.

In that way we could reuse (literally reuse or at least copy-paste) almost all the calling and receiving code on both the `micropython` (`trezor/ui/layouts/<model>/__init__.py`) and `Rust` (`ui/<model>/layout.rs`) side - the only differences would be the `UI` handling on the `Rust` side.

It would mean uniting the strategies mostly for `string` handling - in the `model T` redesign, we tend to mostly pre-process them in `micropython` to avoid working with them in `Rust`, where we only show them. On the other hand, in `model R`, we are mostly processing them in `Rust` - which has its own issues.

As `model T` redesign will be "finished" sooner than `model R`, I am (from the `R` perspective) ready to "copy" anything that we will finalize for `model T` so that both models are as similar as possible.

matejcik, 2023-01-12:
outstanding issue:
`confirm_output_r` is a single flow that allows going back and forth between recipient & amount, no buttonrequest in the middle

`confirm_output` on T2T1 is two `confirm_value` calls each of which sends a buttonrequest.

not sure which way to unify here, an open question is also whether we should allow going back on TT too

matejcik, 2023-01-12:
same thing is true for `confirm_total` where the two screens are total amount + fee

matejcik, 2023-01-12:
T2B1:
* expose `confirm_value` to Python and delegate to `confirm_action` in Rust
* delegate `confirm_joint_total` on Rust side to `confirm_properties`, dtto for `confirm_modify_fee`
* `confirm_modify_output`: delegate to `confirm_properties` (not confirm_action) on Rust side
* `confirm_with_info` and `confirm_more` are missing designs, we will see if we need to implement them separately for T2B1 or delegate to something
* `confirm_coinjoin` can delegate to `confirm_properties` for now
* `confirm_reset_device` - delegate
* `confirm_recovery` - delegate
* `show_remaining_shares` -- only called from info button on `confirm_recovery` so never happens on T2B1. for now let's raise NotImplementedError from Python as it is now, in the future we can raise the exception from Rust if we need to.

matejcik, 2023-01-12:
`show_group_share_success` -- for now delegate to `confirm_action` in Rust, but maybe worth considering merging with `show_popup` on the T2T1 side?

grdddj, 2023-01-17:
> T2B1:
>
>     * expose `confirm_value` to Python and delegate to `confirm_action` in Rust
>
>     * delegate `confirm_joint_total` on Rust side to `confirm_properties`, dtto for `confirm_modify_fee` [...]

All of these things (creating new `trezorui2.xxx` functions for `model R`) done in [757fdf7](https://github.com/trezor/trezor-firmware/pull/2610/commits/757fdf717af17c282b62ceb17bbe48c797c6395d)

Current situation of `Rust API` between models:

**TR-only**
confirm_output ['tr']
confirm_total ['tr']
show_receive_address ['tr']
tutorial ['tr']

**TT-only**
show_qr ['tt']
show_error ['tt']
show_warning ['tt']
show_success ['tt']
show_simple ['tt']
confirm_with_info ['tt']
confirm_more ['tt']
show_remaining_shares ['tt']

**Conflicting implementations**
show_share_words
show_info

grdddj, 2023-01-20:
TR things done in https://github.com/trezor/trezor-firmware/pull/2610

closed 2023-01-20 by grdddj