trezor/trezor-firmware #2442: Fee rate not displayed when replacing transaction

matejcik, 2022-08-08:
**Describe the bug**
#1294 implements fee rate for "sign transaction" flow.
Fee rate should probably be displayed on "modify fee" screen as well.

screenshots:
https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/2826526641/artifacts/test_ui_report/passed/TT_bitcoin-test_signtx_replacement.py::test_p2tr_fee_bump.html

sime, 2022-08-25:
We have the space to include the fee rate?

Concept: https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Trezor-T%2C-Trezor-One?node-id=3539%3A10964

matejcik, 2022-08-25:
this design is for the "total for transaction" screen, not for "modify fee" when replacing. screenshots: https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/2930791419/artifacts/test_ui_report/passed/TT_bitcoin-test_signtx_replacement.py::test_p2pkh_fee_bump.html
relevant screen copied so it doesn't expire again:
[image]

with that said, the same concept should work here -- the screen fits exactly _one_ additional line, so if the fee rate does not have a heading of its own, it will work fine

grdddj, 2022-08-30:
`QA` testing instructions:
- modify fee/replace transaction in `Suite` (or anywhere else) and check whether the `Modify fee` screen shows a fee rate in brackets below transaction fee - for example `(66.4 sat/vB)`

[image]

grdddj, 2022-08-30:
We might consider doing it also for `legacy`, the fee-rate is not shown there when replacing.

There is also one free row left, where we could fit it. Question is whether it is worth the effort...

[image]

matejcik, 2022-08-30:
this issue is also tagged legacy so I believe that applies, yes

grdddj, 2022-08-31:
> this issue is also tagged legacy so I believe that applies, yes

Done in https://github.com/trezor/trezor-firmware/pull/2481

closed 2022-09-12 by grdddj

STew790, 2022-09-20:
**QA OK**

Trezor T:
[image]
Trezor One:
[image]

**Info:**
 - Suite version: web 22.9.2 (cfeba9a3533ec7f4a5d4f31da4ac8c8b768547ed)
 - Browser: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/105.0.0.0 Safari/537.36
 - OS: Linux x86_64
 - Screen: 1920x1080
 - Device: model T 2.5.3 Universal (revision 4bed278e80d23077676128eba8cb2478fcd31120)
 - Device: model 1 1.11.3 Universal (revision 4bed278e80d23077676128eba8cb2478fcd31120)
 - Transport: bridge 2.0.32