trezor/trezor-firmware #2453: Monero primary address discrepancy between app and device

bosomt, 2022-08-17:
**Describe the bug**
I'm not sure if its expected or not or if it's problem with moneroGUI or Trezor.
But i was not able to replicate it with 2.5.1 FW.
When i press show address on device i see completely different address.
But it belongs to same account /i just sent some Monero to that address and i still have same amount of xmr.

When you create secondary address it works as expected. Device and app address does match, looks like only Primary address is affected.

**Firmware version and revision**

**QA NOK** Device: model T 2.5.2 regular (revision 0d87b55ba4fed7eecc72bf2a94ee473830b095e9)
**QA OK** Device: model T 2.5.1 regular (revision 85a26d2c9593bcdf858c2d718d79951ca927a0c3)

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: moneroGUI 0.18.1.0-release (Qt 5.12.8)
 - OS: macARM

**To Reproduce**
Steps to reproduce the behavior:
1. Go to moneroGUI/Receive
2. Click on `Shonw on device`
3. Check address on device display and confirm
5. Compare with address displayed in app

**Screenshots**
[image]

[image]

[image]

matejcik, 2022-08-17:
@ph4r05 please take a look.
i vaguely remember something about unified addresses, could this be it?

ph4r05, 2022-08-18:
Interesting! Thanks for the report. Unfortunately, I am quite busy till 11.9.2022 so I am not sure I will find time to take a look at this till then.

matejcik, 2022-08-25:
@selsta since you are involved here, anything you can tell us about this issue from a quick look?
something something integrated address?

selsta, 2022-08-25:
This does not seem related to integrated addresses. The primary address is a special case, it seems the firmware simply calculated `subaddress(0, 0)` instead of displaying the primary address. Should be easy to fix.

ph4r05, 2022-08-31:
Thanks for the diagnosis, @selsta. It seems to be correct, If subaddress indices are `(0, 0)`, primary address should be displayed. 

@matejcik this code is on the blame https://github.com/trezor/trezor-firmware/blob/33c174491f45d11bbc9587e953f09b95ef834657/core/src/apps/monero/get_address.py#L48-L60

PR #2478 fixes it

closed 2022-09-09 by matejcik

bosomt, 2022-09-20:
@ph4r05 looks like we need new moneroGUI version to test this fix, im not able to open or create new wallet.

`monero-gui-mac-x64-v0.18.1.1`

```
2022-09-20 18:45:21.983	E Get public address exception: Call method failed
2022-09-20 18:45:21.985	E Cannot get a device address
```

[image]

selsta, 2022-09-20:
This _should_ be a purely firmware side issue, I don't see why a new GUI version should be required here.

bosomt, 2022-09-20:
Yes I have but new version `Version: 2.0.32`, as soon i changed it to `Version: 2.0.31` and disabled PIN code im able to connect to device.

bosomt, 2022-09-20:
**QA OK**

addresses works as expected 

 - Device: model T 2.5.3 Universal (revision 4bed278e80d23077676128eba8cb2478fcd31120)
 - Transport: bridge 2.0.31

selsta, 2022-09-20:
might make sense to look into why bridge version 2.0.32 fails

bosomt, 2022-09-21:
Works as expected today with 2.0.32 🤷🏿‍♂️ 

[image]

commit a697f29e (2022-09-01): fix(xmr): fixes #2453, incorrect address display
PR #2478 (2022-09-09): fix(xmr): fixes #2453, incorrect address display
Fixes #2453
commit 2fe40f70 (2022-09-09): fix(xmr): fixes #2453, incorrect address display